OpenVoiceOS/ovos-buildroot #5: Suggestion/Buildroot: Addition of a mic array kernel drivers

JarbasAl, 2018-11-01:
Hi, i think MycroftOS is a great idea!

Since it is aimed at raspi3 for now, it would make sense to support some common mic arrays, 

Maybe also support some other hats that make sense for mycroft

Suggestions:
- respeaker, i see you have plans for it already #1 
- matrix voice
- google AIY kits (vision and voice)
- sense hat, not sure how much sense this one makes, im making a mycroft enclosure for it

j1nx, 2018-11-01:
You are definitely right! Buildroot should take care of all the drivers and OS configuration. The visuals should be handled by a skill I think.

I already have a ReSpeaker 4-Mic array at hand, so that one is indeed on #1 but agree that at this point in time AIY and Matrix voice are the next two important ones.

The new ReSpeaker arrays are on this list as well, but I believe they use the exact same driver, just some other OS configuration files. The only "worry" I have at the moment is to properly figure out at boot which one the user has. However that are future concerns.

j1nx, 2018-11-03:
ReSpeaker kernel drivers and OS configuration is merged. At this point we do not have configuration wizards yet, but later on we will have some sort of cornfiguration wizard where you could select it and it will be used. Similar as the picroft, but I do not want that to be done over the cli.

In the future I would like to do that webbased. A small local webserver and easy setup of hardware and such.

j1nx, 2024-01-24:
At this moment we support the following microphones

- USB mic (PS3 Eye and/ or JABRA as example)
- WM8960 2mic based HAT's (respeaker,waveshare, etc)
- 4mic square ReSpeaker
- 4 mic lineair ReSpeaker
- 6 mic square ReSpeaker
- Mark1 (rpi-proto based)
- Mark2 Dev-Kit (SJ201 Rev6 based)
- Mark2 retail (SJ201 V10 based)
- Google AIY kit

Going forward, we might include more if time allows and hardware allows. For now closing this as complete

closed 2024-01-24 by j1nx