OpenVoiceOS/ovos-buildroot #18: Fix / Check all Buildroot package dependencies

j1nx, 2019-01-30:
### Information
All packages of buildroot-external are there and by using the provided defconfig file(s) for building the firmware, everything will be properly downloaded and build. 

However...

I have not yet been putting efforts in linking all the right dependencies for each package. Selecting one, should selecting some others and/or building some before the other. In the host or the target etc.

This cleanup should happen eventually.

denics, 2023-02-04:
## List of packages to check

- [ ] armnn
- [ ] baresip
- [ ] btspeaker
- [ ] clog
- [ ] compute-library
- [ ] cpuinfo
- [ ] deepfilternet-ladspa
- [ ] docbook-xml
- [ ] docbook-xsl
- [ ] fann
- [ ] farmhash
- [ ] fft2d
- [ ] fp16
- [ ] fxdiv
- [ ] gemmlowp
- [ ] growdisk-service
- [ ] hdf5
- [ ] hostname-service
- [ ] kaldi
- [ ] kf5-attica
- [ ] kf5-breeze-icons
- [ ] kf5-kactivities
- [ ] kf5-kactivities-stats
- [ ] kf5-karchive
- [ ] kf5-kauth
- [ ] kf5-kbookmarks
- [ ] kf5-kcmutils
- [ ] kf5-kcodecs
- [ ] kf5-kcompletion
- [ ] kf5-kconfig
- [ ] kf5-kconfigwidgets
- [ ] kf5-kcrash
- [ ] kf5-kdbusaddons
- [ ] kf5-kdeclarative
- [ ] kf5-kdeconnect
- [ ] kf5-kdecoration
- [ ] kf5-kded
- [ ] kf5-kdelibs4support
- [ ] kf5-kdesignerplugin
- [ ] kf5-kdesu
- [ ] kf5-kdoctools
- [ ] kf5-kemoticons
- [ ] kf5-kglobalaccel
- [ ] kf5-kguiaddons
- [ ] kf5-ki18n
- [ ] kf5-kiconthemes
- [ ] kf5-kidletime
- [ ] kf5-kinit
- [ ] kf5-kio
- [ ] kf5-kirigami2
- [ ] kf5-kitemmodels
- [ ] kf5-kitemviews
- [ ] kf5-kjobwidgets
- [ ] kf5-knewstuff
- [ ] kf5-knotifications
- [ ] kf5-knotifyconfig
- [ ] kf5-kpackage
- [ ] kf5-kparts
- [ ] kf5-kpeople
- [ ] kf5-kpty
- [ ] kf5-krunner
- [ ] kf5-kscreen
- [ ] kf5-kscreenlocker
- [ ] kf5-kservice
- [ ] kf5-ksysguard
- [ ] kf5-ktextwidgets
- [ ] kf5-kunitconversion
- [ ] kf5-kwallet
[... 287 lines cut ...]
- [ ] skill-ovos-notes
- [ ] skill-ovos-personal
- [ ] skill-ovos-settings
- [ ] skill-ovos-setup
- [ ] skill-ovos-stop
- [ ] skill-ovos-timer
- [ ] skill-ovos-volume
- [ ] skill-ovos-weather
- [ ] skill-ovos-wolfie
- [ ] skill-youtube-music
- [ ] snapcast
- [ ] spotifyd
- [ ] tensorflow-lite
- [ ] userland-tools
- [ ] virtual-touch
- [ ] vnc-eglfs
- [ ] vocalfusion
- [ ] vosk-api
- [ ] whisper-tflite
- [ ] whispercpp
- [ ] wifi-connect
- [ ] wiringpi2
- [ ] xnnpack
- [ ] xorgproto-mesafix
- [ ] xvf3510

j1nx, 2024-01-24:
As we are now utilizing the containerized system, we should go for a good cleanup within the external buildroot tree.

From there on, what is left might still need to be checked.