OpenVoiceOS/ovos-buildroot #19: Make sure all Skills from the market can install (dependencies)

j1nx, 2019-01-30:
### Information
Asd there are no GCC / Compile tools available on MycroftOS, we have to make sure all Skills from the Market can be installed without any problems (unless the skill is very specific for a certain platform, but even then I will try to get it done somehow)

Python3.7 and PIP are installed. PIP should be configured to also look at piwheels (already there). Where possible wheels should be preferred otherwise source should be ok, but sources with gcc (or other compiler) should be wheels only and if that is not be possible be pre-installed and therefor compiled at buildtime of the system.

ChanceNCounter, 2021-07-10:
Should this be moved to the installer skill?

j1nx, 2021-07-10:
This can be closed. Had a look at this and included most of the major gcc/python pip packages that might be needed.

closed 2021-07-10 by j1nx