OpenVoiceOS/ovos-buildroot #20: Implement webbased backend

j1nx, 2019-01-30:
### Information
MycroftOS is a voice controlled system, however some settings are way easier to configure via a webbased system. Hence Mycroft A.I. even has exactly this with; home.mycroft.ai

I strongly believe MycroftOS should also have some sort of webbased backend where you can change stuff. Not only pairing and skill settings, but also hooks into the new to be released GUI part. So a bullit list of what I think in the end should be available via a browser connecting to MycroftOS.
- CLI, a simple webversion os mycroft-cli-client
- GUI output (Although, I am not sure if all the QT stuff plays nice or how to have this done)
- Option to use the backend for pairing and skill settings/api's (https://github.com/MycroftAI/personal-backend)
- Option to record and create custom wake words similar as; https://github.com/bjoernkarmann/project_alias
- Options to change volume controls, mute, etc (https://github.com/j1nx/PaWebControl)
- Any other system options/settings, such as network, lan, wifi, hdmi output, etc.

I believe most of the appraoches I have seen and linked above use npm/flask/etc. So believe this should be the way forward to minimise dependencies and extra packages. (Wifi-Wizard also uses it, so npm should be there rather sooner then later)

JarbasAl, 2020-11-18:
quick note, i stopped maintaining personal-backend when selene was released, some more updated have meanwhile been made at https://github.com/OpenJarbas/mock-backend

i also package this as a skill, anyone can simply install it and not require backend https://github.com/JarbasSkills/skill-mock-backend

i feel the settings etc should be essentially handled by the GUI https://github.com/MycroftAI/mycroft-core/pull/2698

a local frontend might also be served for headless devices if needed, this could happen at/integrate with mock-backend

if there is interest i can move mock-backend under OVOS organization

j1nx, 2024-01-24:
Old issue tracker for MycroftOS. We now do have some webbased systems under the organsation, however this has been moved up one level.

closed 2024-01-24 by j1nx