OpenVoiceOS/ovos-buildroot #30: Connection rejected

eduardojj, 2019-12-10:
Hello,
I'm having problems connecting to api.mycroft.ai it keeps sending connection rejected error, even in mycroft-cli-client.
I'm still trying to figure out if it needs to be connected to a different network besides the ssh with my host.
Can you help?

j1nx, 2019-12-10:
Right, no problem! Did you compiled the image yourself or did you downloaded one of the alpha's?

If you can SSH into the box with mycroft/mycroft it means that you do have network connected. Let's start with testing if you have Internet access on the device.

SSH into the device. Cancel the mycroft-cli-client is started and run a simple "ping google.com"

eduardojj, 2019-12-10:
I compiled the image myself for a project at university that is constrained to buildroot.

I ssh into the rasp via `ssh root@ipaddress`
Here's a ping test
`eduardo@X550JK:~/mycroft-core/bin$ ssh root@10.42.0.245
root@10.42.0.245's password:
ping google.com
PING google.com (216.58.211.46): 56 data bytes
64 bytes from 216.58.211.46: seq=0 ttl=54 time=24.242 ms
64 bytes from 216.58.211.46: seq=1 ttl=54 time=23.378 ms
64 bytes from 216.58.211.46: seq=2 ttl=54 time=23.759 ms
64 bytes from 216.58.211.46: seq=3 ttl=54 time=24.358 ms
64 bytes from 216.58.211.46: seq=4 ttl=54 time=72.176 ms
64 bytes from 216.58.211.46: seq=5 ttl=54 time=21.608 ms
64 bytes from 216.58.211.46: seq=6 ttl=54 time=253.258 ms
^Z
[1]+  Stopped(SIGTSTP)        ping google.com
`
I can use mycroft-audio-test, but on the others i get connection rejected
Thx for the help and quick reply.

j1nx, 2019-12-10:
Hmmm, strange that you can login with root as that is disabled already a while ago.

Which version are you currently running? If you have a look at 'git log' within the folder, what is the latest commit? I hope you checked out the develop branch, because master is running far behind, which i did not want to update/merge before I reach a certain beta status.

eduardojj, 2019-12-10:
I was using master, i'll try with the development branch

closed 2019-12-10 by eduardojj

j1nx, 2019-12-10:
Right, no problem. Don't forget to pull in all the latest and greatest code. Than in the folder run;

make clean
make rpi3-config
make rpi3

Get some coffee or two, three, four ....

j1nx, 2019-12-10:
Login with the user/pass: mycroft/mycroft
At first boot, the image get's expanded over the SD card and Mycroft starts downloading a bunch of package when it has LAN connection. It might take a few minutes before 'mycroft-cli-client' runs.

I pushed a bunch of changes which I have not fully tested myself. A fresh build is still compiling as we speak, so please report back both positive as negative.

eduardojj, 2019-12-10:
Will do.
Thank you very much for your time

eduardojj, 2019-12-10:
I'm sorry, but the buildroot make seems to be taking to long to fetch all references for linux-headers. It's kinda stuck in "Fetching all references"

j1nx, 2019-12-10:
Yes, you are building a complete linux system from scratch and it pull in A LOT of source code. Especially a new build will take a long time, however rebuilding it should be faster as all packages are already downloaded. The build system also uses ccach, so next time it should be a bit faser.

But overall, for me using a simple build machine It takes all night for a fresh build.

I have a new alpha_4 build available. Will upload it tonight and share the code for you if you would like to try out a new build.

j1nx, 2019-12-10:
https://www.j1nx.nl/downloads/mycroftos-0-1-0-alpha4-rpi3/