OpenVoiceOS/ovos-buildroot #31: Make error in mpv linking

eduardojj, 2019-12-10:
So I left the PC compiling buildroot and all and i got a error when linking mpv package
It goes like this:

`mpv 0.29.1 Building`

`cd /home/eduardo/buildroot/MycroftOS/buildroot/output/build/mpv-0.29.1/ && PATH="/home/eduardo/buildroot/MycroftOS/buildroot/output/host/bin:/home/eduardo/buildroot/MycroftOS/buildroot/output/host/sbin:/home/eduardo/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" /home/eduardo/buildroot/MycroftOS/buildroot/output/host/bin/python2 /home/eduardo/buildroot/MycroftOS/buildroot/output/host/bin/waf build -j 9 ` 

`Waf: Entering directory /home/eduardo/buildroot/MycroftOS/buildroot/output/build/mpv-0.29.1/build'`
`[1/1] Compiling version.sh `
`[208/208] Linking build/mpv`
`/home/eduardo/buildroot/MycroftOS/buildroot/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/8.3.0/../../../../arm-buildroot-linux-gnueabihf/bin/ld: video/out/opengl/context_rpi.c.19.o: undefined reference to symbol 'vc_dispmanx_display_close'`
`/home/eduardo/buildroot/MycroftOS/buildroot/output/host/lib/gcc/arm-buildroot-linux-gnueabihf/8.3.0/../../../../arm-buildroot-linux-gnueabihf/bin/ld: /home/eduardo/buildroot/MycroftOS/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/lib/libvchostif.so: error adding symbols: DSO missing from command line`
`collect2: error: ld returned 1 exit status`

`Waf: Leaving directory /home/eduardo/buildroot/MycroftOS/buildroot/output/build/mpv-0.29.1/build'`
`Build failed`
` -> task in 'mpv' failed with exit status 1 (run with -v to display more information)`
`make[1]: *** [package/pkg-generic.mk:241: /home/eduardo/buildroot/MycroftOS/buildroot/output/build/mpv-0.29.1/.stamp_built] Error 1`
`make[1]: Leaving directory '/home/eduardo/buildroot/MycroftOS/buildroot'
make: *** [Makefile:25: rpi3] Error 1`

Thank you

j1nx, 2019-12-10:
Ah, my apologies, we forgot to patch Buildroot. It is in the README, but I am sure you just took my advise with the make commands. I didn't think about it as well.

I am afraid we need to start from scratch.

https://github.com/j1nx/MycroftOS/blob/develop/documentation/building.md

eduardojj, 2019-12-10:
For now i disabled MPV since for my application i wont be needing it, but ill dig deeper into this later.

j1nx, 2019-12-10:
Fine, but there are a whole lot of patches that need to be applied before you start the compile. Otherwise you end up with complete different software version for quite some packages.

It is mandatory that you run the command;

./scripts/br-patches.sh

Best is to stop this compile run and restart from scratch because I also thing your Buildroot is not yet up to date to the last 2019.11.x version.

To make sure everything goes as planned best is to delete the MycroftOS folder and run the commands from a to z in above linked documentation.

j1nx, 2019-12-10:
And before I forget;

cd ~/
git clone --recurse-submodules https://github.com/j1nx/MycroftOS.git
cd MycroftOS
**git checkout develop**

Than patch Buildroot by;

./scripts/br-patches.sh

Than we can start tjhe build with
make rpi3-config
make rpi3

Go to sleep. At the very end it error's out because of SpeechRegognition, I still need to solve that but it only happens at a clean run so very annoying to test.

Anyhow, if it errors out on "SpeechRegognition", just restart the compile by running;

make rpi3

It will continues where it left...

closed 2019-12-10 by eduardojj

eduardojj, 2019-12-10:
I started from scratch and when it was writing the image it stop due to a error. On further inspection after running `make -C buildroot BR2_EXTERNAL=buildroot-external 2>&1` I noticed i had a Kconfig fragment error on busybox.
Any idea?
Here's the output:
`for f in buildroot-external/busybox.config ; do if [ ! -f "${f}" ]; then printf "Kconfig fragment '%s' for '%s' does not exist\n" "${f}" "busybox"; exit 1; fi; done`
`Kconfig fragment 'buildroot-external/busybox.config' for 'busybox' does not exist`