OpenVoiceOS/ovos-buildroot #32: mycroft-cli-client associeted errors

eduardojj, 2019-12-11:
I've been having trouble ending the Make but i bypassed it and finished it, it was giving a error when it was runninh the post-image script, then i loaded the image and it runned perfectly until i tried to start mycorft-cli-client it gave the error of unsuported locale
`mycroft@MycroftOS:~ $ mycroft-cli-client`
`Traceback (most recent call last):`
 ` File "/bin/mycroft-cli-client", line 11, in <module>`
 `   load_entry_point('mycroft-core==19.8.4', 'console_scripts', 'mycroft-cli-client')()`
 ` File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)`
 ` File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()`
 ` File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()`
 ` File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)`
 ` File "/usr/lib/python3.8/site-packages/mycroft/client/text/__main__.py", line 21, in <module>
    from .text_client import (`
 ` File "/usr/lib/python3.8/site-packages/mycroft/client/text/text_client.py", line 39, in <module>
    locale.setlocale(locale.LC_ALL, "")  # Set LC_ALL to user default`
 ` File "/usr/lib/python3.8/locale.py", line 608, in setlocale
    return _setlocale(category, locale)`
`locale.Error: unsupported locale setting`
I fixed it by exporting the variable LC_ALL to a locale and it started normaly, but i'm getting a unhandled expression inside the client. This was all after i registered the device at home.mycroft.ai.
Thank you for your time.

eduardojj, 2019-12-11:
This was already bypassed, not solved, but the issue is after i register the device its fine and works the only problem is the volume skill and audio output, it can seem to find the audio output correctly and i dont hear the answer, but if i write it i get the answer correctly. I can see the microphone volume change in the client and it receives the wake word and query, but when i use mycroft-audio-test i dont hear a thing.

j1nx, 2019-12-11:
Right, back on my feet...

First things first. LC_ALL should be set by the buildsystem to "en_US.UTF-8"
https://github.com/j1nx/MycroftOS/blob/develop/buildroot-external/configs/rpi3_defconfig#L22

What did you do exactly? Than I can double check on this side and if needed script the same.

Secondly about the possible Python errors when you start "mycroft-cli-client" a bit to soon after the first boot. That is a known thing. Starting the Mycroft A.I. software stack for the first time means it is downloading all default skills and such. During that it also downloads and installs a lot of additional needed Python packages. If you would give it a bit more time, the mycroft-cli-client starts as wanted without errors.

Anyhow: Last but not least. The sound output. Funny, that sounds excactly like the topic here;
https://community.mycroft.ai/t/mycroft-with-respeaker-2-mic/7635

I am looking into it and also requested that user to give the alpha4 image a good, to see if we can hunt down that bug...

eduardojj, 2019-12-12:
Great news, those problems with the make probably my fault, sorry about that, because i made an image yesterday and the only proble was on speechrecognition but thats a known bug, and the audio problem was solved by copying a configuration.json file to a correct location, which i suspect was due to the failed make, sorry i dont know the details because my group collegue did that fix, but if you want i can try.

j1nx, 2019-12-12:
Perfect! Will await your feedback about the LOCALE. With a bit of luck, that has resolved itself as well.

eduardojj, 2019-12-16:
The locale is no longer a problem. The problems that now we have is sporadic reboots with no associated trigger.
We have also a question, is it possible to remove the startup and shutdown Mycroft images?

j1nx, 2019-12-16:
ATM I also witnessed strange things with the last Buildroot 2019.11.x upgrade. Working on a downgraded version as we speak. Untill Python 3.8 is more stable, I stick with the 2019.08.3 stable release of Buildroot.

You will most likely see the commits come in in the development branch later today or tomorrow.

To answer your question about the splash screens. Sure, you can turn off the service via menuconfig. Just run this command before you run the make command, after the "make clean " and "make rpi3-config" command;

`make -C buildroot BR2_EXTERNAL=../buildroot-external menuconfig`

Than in the ncurses menu go to the bottom section, "External packages" -> "Mycroft A.I. Personal Assistant" and deselect the Mycroft Splash package. Exit back out of this menu and then you can run the build again; "make rpi3"

j1nx, 2020-01-06:
Fixed by 4061037d0bb69b9c23f4b154674b22482c813ddf

closed 2020-01-06 by j1nx