OpenVoiceOS/ovos-buildroot #33: The new version.

czadikem, 2020-05-31:
Hi @j1nx I was wondering if their was a way that I could get your latest relase MycroftOS-GUI_0.1.0_rpi3_alpha8?  Thanks

j1nx, 2020-06-01:
You can always build it yourself, but it takes quite some hours depending on your machine. Have a look at the documentation;
https://github.com/j1nx/MycroftOS/blob/develop/documentation/building.md

Otherwise, just a bit more patience. Getting there, but there are always unforeseen nast little crappy bugs that prevent the image from being fully dummy userproof.

czadikem, 2020-06-01:
Thank you for the help.

j1nx, 2020-06-01:
I just pushed some updates to the development that is try-run-building as we speak. There might be some erroring out as I have not fully teated the build yet.

Will report back if all is fine.

czadikem, 2020-06-01:
OK thank you

poveteen, 2020-06-21:
I'm very excited to try Alpha 8. It’s awesome.
I want to try MycroftOS Alpha8.
I successfully built "rpi_64-gui". However, the Speaker 4-Mic Array microphone was not recognized.
So I built "rpi4-gui", but in Ubuntu 20.4, I had an error compiling the seeed-voicecard as shown below.

Is there a solution?
Thank you.

====================================
make[1]: Leaving directory '/home/mycroftos/MycroftOS/buildroot'
build rpi4-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external 2>&1 | tee logs/buildroot_rpi4-gui_output.txt
make[1]: Entering directory '/home/mycroftos/MycroftOS/buildroot'
/usr/bin/make -j1 O=/home/mycroftos/MycroftOS/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
>>> respeaker 6b4133b5f056fd0218f3ffd197df45fab2ae1ef7 Building
>>> respeaker 6b4133b5f056fd0218f3ffd197df45fab2ae1ef7 Building kernel module(s)
PATH="/home/mycroftos/MycroftOS/buildroot/output/host/bin:/home/mycroftos/MycroftOS/buildroot/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PKG_CONFIG="/home/mycroftos/MycroftOS/buildroot/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/mycroftos/MycroftOS/buildroot/output/host/lib/pkgconfig:/home/mycroftos/MycroftOS/buildroot/output/host/share/pkgconfig" BR_BINARIES_DIR=/home/mycroftos/MycroftOS/buildroot/output/images KCFLAGS=-Wno-attribute-alias DTC_FLAGS=-@ /usr/bin/make -j8 -C /home/mycroftos/MycroftOS/buildroot/output/build/linux-custom HOSTCC="/home/mycroftos/MycroftOS/buildroot/output/host/bin/ccache /usr/bin/gcc -O2 -I/home/mycroftos/MycroftOS/buildroot/output/host/include -L/home/mycroftos/MycroftOS/buildroot/output/host/lib -Wl,-rpath,/home/mycroftos/MycroftOS/buildroot/output/host/lib" ARCH=arm INSTALL_MOD_PATH=/home/mycroftos/MycroftOS/buildroot/output/target CROSS_COMPILE="/home/mycroftos/MycroftOS/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-" DEPMOD=/home/mycroftos/MycroftOS/buildroot/output/host/sbin/depmod INSTALL_MOD_STRIP=1  PWD=/home/mycroftos/MycroftOS/buildroot/output/build/respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/. M=/home/mycroftos/MycroftOS/buildroot/output/build/respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/. modules
  CC [M]  /home/mycroftos/MycroftOS/buildroot/output/build/respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/./seeed-voicecard.o
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c: In function ‘seeed_voice_card_dai_init’:
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:262:8: error: implicit declaration of function ‘asoc_simple_card_init_dai’; did you mean ‘asoc_simple_dai_init’? [-Werror=implicit-function-declaration]
  262 |  ret = asoc_simple_card_init_dai(codec, &dai_props->codec_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_dai_init
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c: In function ‘seeed_voice_card_dai_link_of’:
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:315:8: error: implicit declaration of function ‘asoc_simple_card_parse_daifmt’; did you mean ‘asoc_simple_parse_daifmt’? [-Werror=implicit-function-declaration]
  315 |  ret = asoc_simple_card_parse_daifmt(dev, node, codec,
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_daifmt
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:322:8: error: implicit declaration of function ‘asoc_simple_card_parse_cpu’; did you mean ‘asoc_simple_parse_cpu’? [-Werror=implicit-function-declaration]
  322 |  ret = asoc_simple_card_parse_cpu(cpu, dai_link,
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_cpu
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:328:8: error: implicit declaration of function ‘asoc_simple_card_parse_codec’; did you mean ‘asoc_simple_parse_codec’? [-Werror=implicit-function-declaration]
  328 |  ret = asoc_simple_card_parse_codec(codec, dai_link, DAI, CELL);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_codec
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:340:8: error: implicit declaration of function ‘asoc_simple_card_parse_platform’; did you mean ‘asoc_simple_parse_platform’? [-Werror=implicit-function-declaration]
  340 |  ret = asoc_simple_card_parse_platform(plat, dai_link, DAI, CELL);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_platform
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:365:8: error: implicit declaration of function ‘asoc_simple_card_parse_clk_cpu’; did you mean ‘asoc_simple_parse_clk_cpu’? [-Werror=implicit-function-declaration]
  365 |  ret = asoc_simple_card_parse_clk_cpu(dev, cpu, dai_link, cpu_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk_cpu
../respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/seeed-voicecard.c:373:8: error: implicit declaration of function ‘asoc_simple_card_parse_clk_codec’; did you mean ‘asoc_simple_parse_clk_codec’? [-Werror=implicit-function-declaration]
  373 |  ret = asoc_simple_card_parse_clk_codec(dev, codec, dai_link, codec_dai);
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        asoc_simple_parse_clk_codec
[... cut ...]
make[4]: *** [scripts/Makefile.build:266: /home/mycroftos/MycroftOS/buildroot/output/build/respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/./seeed-voicecard.o] Error 1
make[3]: *** [Makefile:1702: /home/mycroftos/MycroftOS/buildroot/output/build/respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/.] Error 2
make[2]: *** [package/pkg-generic.mk:270: /home/mycroftos/MycroftOS/buildroot/output/build/respeaker-6b4133b5f056fd0218f3ffd197df45fab2ae1ef7/.stamp_built] Error 2
make[1]: *** [Makefile:84: _all] Error 2

j1nx, 2020-07-23:
A new alpha8 build has just been uploaded here;
https://www.j1nx.nl/downloads/

A rpi3 version is currently being build and uploaded and will be available on the same page a bit later. If all works a bit as expected, the develop branche will be merged into master soon and be tagged as alpha8 release.

closed 2020-07-23 by j1nx