OpenVoiceOS/ovos-buildroot #34: Can't download your img file

czadikem, 2020-07-23:
When I download your image file it says 0b can you give me another link please. the link I used was https://www.j1nx.nl/download/mycroftos-0-1-0-alpha8-rpi4-64bit/

Thanks

czadikem, 2020-07-23:
Here 
[image]
is a picture of me using commandline to try and download the file

j1nx, 2020-07-24:
And this direct link?
https://j1nx.stackstorage.com/s/dUnc2C50fEv73La3

closed 2020-07-28 by j1nx

czadikem, 2020-07-28:
Sorry for the late reply.  But The link above is not working.  So I would
change your file storage to something like Google drive etc.

j1nx, 2020-07-29:
https://drive.google.com/file/d/1So5knzMwUtrqnjUE2iJAgt5yS9nhJ1ME/view?usp=sharing

Give it a try.

czadikem, 2020-07-31:
Thanks that works for me😄!