OpenVoiceOS/ovos-buildroot #36: Setting up Wifi fails

JoergZ2, 2020-09-07:
Hi Peter, we had a chat on chat.mycroft.ai and you told me how to setup wifi. But it didn't work. MycroftOS opens a Mycroft SSID which I can connect. Then I can see my SSIDs and choose one of them and put in the password but then nothing happens. Is there something additional to do?

j1nx, 2020-09-07:
Nope, that sounds about right. Anyhow, you just reached the point of where I am currently at in the development.

The network stack used in the image is NetworkManager. Maybe as quick fix you can use the nmcli command at the linux cli to get you going for now. A next release have it finished and fixed. The image you got is actually an in between officially not yet released version 😉

JoergZ2, 2020-09-08:
> Maybe as quick fix you can use the nmcli command at the linux cli to get you going for now.

That worked well. Thanks!

j1nx, 2021-05-05:
With the new wifi setup process this Issue is also fixed

closed 2021-05-05 by j1nx