OpenVoiceOS/ovos-buildroot #37: Install fails when 1.1.1.1 is down

hieronymousch, 2020-09-09:
IP is hard coded in /etc/systemd/system/multi-user.target.wants/mycroft.service and was unreachable for me... so installation hung. Maybe ping a DNS name?

j1nx, 2020-09-09:
Hi, them you have a rather old version running as the internet check has been removed a while ago.

Which RPI do you have? So i can give you a link to the latest download available.

hieronymousch, 2020-09-09:
Pi 3B, I just downloaded the release on this page.. Thanks for your help!

hieronymousch, 2020-09-09:
Found the download page via the forum, will try that version. (link: https://www.j1nx.nl/downloads/ )
thanks

j1nx, 2020-09-09:
If you have an RPI4, you are beter of downloading this;
https://j1nx.stackstorage.com/s/LPPpy2VkyV8Vww4I