OpenVoiceOS/ovos-buildroot #38: Can't install home assistant skill

TheLogan, 2020-09-16:
Hi, I'm fairly new to linux and Mycroft and such, but I've got MycroftOS running on a RPi3, and now I'm in the process of trying to install home assistant too, but I've run into a problem. It keeps telling me that: _An error occurred while installing the Python dependencies for homassistant_, I've attempted it with picroft before where it just installed with no issues, and I'm not entirely certain what is wrong here.

TheLogan, 2020-09-16:
I just got the idea to ssh into the device which allowed me to see the logs, and here's what's going on
_ERROR: responses 0.12.0 has requirement urllib3>=1.25.10, but you'll have urllib3 1.24.2 which is incompatible.
ERROR: Command errored out with exit status 1:_

Can I safely update the urllib version?

j1nx, 2020-09-16:
That is strange!

I have a patch in place that downgrades urllib3 to 1.24.2 
https://github.com/j1nx/MycroftOS/blob/develop/buildroot-patches/0012-Downgrade-python-urllib3-to-1.24.2.patch
which ussually means that was a hard requirement for one of the very strict requirements of mycroft itself.

Perhaps the restriction on this one has been loosen up somewhere over time which I did not picked up yet. Will see if I can loosen up the requirement and remove the patch. That should fix this issue.

Will get back to you asap

TheLogan, 2020-09-16:
I wouldn't know, but since the Home assistant skill requires it, I'd assume it to be the case ^_^
Will be happy to test it when you've fiddled with it! :)

j1nx, 2020-09-19:
OK, looked into it. First of all, the HomeAssistant skill needs this PR;
https://github.com/MycroftAI/mycroft-homeassistant/pull/45

Then you need a newer version of MycroftOS after this commit;
https://github.com/j1nx/MycroftOS/commit/f12d6fa69d6e15f88be8b087cc7fe1f9a36eeae7

The latter I can provide, but we have to wait till the PR get's accepted.

TheLogan, 2020-09-29:
Seems to still be open on their end, is there a way to get an update once it's accepted do you know?

stratus-ss, 2020-10-10:
approved and merged PR 45 from Mycroft side.

closed 2021-05-05 by j1nx