OpenVoiceOS/ovos-buildroot #45: Build failures

EVMakers, 2021-01-28:
Can you publish a list of dependencies needed and linux version needed for the build environment. I seem to be in some dependency hell for all device types. rpi4_64-gui errors on gles qt5 error.

make rpi4_64 results in:

**Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (8192 blocks): done
Copying files into the device: __populate_fs: Could not allocate block in ext2 filesystem while writing file "vlc.mo"
mkfs.ext4: Could not allocate block in ext2 filesystem while populating file system
*** Maybe you need to increase the filesystem size (BR2_TARGET_ROOTFS_EXT2_SIZE)
make[2]: *** [fs/ext2/ext2.mk:63: /home/osboxes/source/MycroftOS/buildroot/output/images/rootfs.ext2] Error 1
make[1]: *** [Makefile:84: _all] Error 2
make[1]: Leaving directory '/home/osboxes/source/MycroftOS/buildroot'
cp -f buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64.img
cp: cannot stat 'buildroot/output/images/sdcard.img': No such file or directory**

make rpi4_64-gui results in:

**_CMake Error at src/declarativeimports/core/CMakeLists.txt:29 (add_library):
  Target "corebindingsplugin" links to target "Qt5::Gui_GLESv2" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?_**

I feel like I am very close to success here. Thank you for any help.

j1nx, 2021-01-29:
First things first: Forget about "rpi4_64" as that config is old and will be deleted. Most disk size of the image comes from all the Python packages, so there is not much difference image size wise between the GUI and non-GUI version. Therefor we decided to disregard the non-GUI configuration going forward. So...

You are indeed very close and what you see is not your fault.

At the moment there is exactly the bug that you described for us to fix, where buildroot does not properly detect the GLESv2 libs. However this is a fix for Buildroot itself, and I have not yet learned how to fix it. It is on the list to be fixed straight after the upcoming release so we can start looking at CI/CD systems.

Anyhow;

Build the "rpi4_64-gui" version till it errors out with about GLESv2 lib.

Now edit the following file;
`buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake`
At the bottom of the file you will find this line;
`_qt5gui_find_extra_libs(OPENGL "GLES" "" "")`
Change it to;
`_qt5gui_find_extra_libs(OPENGL "**/home/peter/Development/OpenVoiceOS**/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/libGLESv2.so" "" "**/home/peter/Development/OpenVoiceOS**/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/include/libdrm")`
Of course make sure you change "the above **/home/peter/Development/OpenVoiceOS** to your own path.

You can then continue the build proces with just rerunning the make rpi4_64-gui

EVMakers, 2021-01-30:
Success! Build running on Raspi4. Thank you. Should I point to your github as a custom skill repo?

j1nx, 2021-01-30:
Perfect. We are in the middle of some changes for sound and expanding the sdcard partition. So if any og those bite you, you might be beter off going e few commits back.

closed 2021-01-30 by j1nx