OpenVoiceOS/ovos-buildroot #46: Can't build project

TheLogan, 2021-02-03:
I've been trying to build the project today and I get as far as running the make script (something I'm quite unfamiliar with), I've got everything I'm doing below, I'm using GIT Bash window in Windows 10

```
Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation
$ git clone --recurse-submodules https://github.com/j1nx/MycroftOS.git
Cloning into 'MycroftOS'...
remote: Enumerating objects: 425, done.
remote: Counting objects: 100% (425/425), done.
remote: Compressing objects: 100% (273/273), done.
remote: Total 7631 (delta 215), reused 313 (delta 133), pack-reused 7206
Receiving objects: 100% (7631/7631), 324.26 MiB | 25.27 MiB/s, done.
Resolving deltas: 100% (3529/3529), done.
Checking out files: 100% (1577/1577), done.
Submodule 'buildroot' (https://github.com/buildroot/buildroot.git) registered for path 'buildroot'
Cloning into 'C:/Home automation/MycroftOS/buildroot'...
remote: Enumerating objects: 270, done.
remote: Counting objects: 100% (270/270), done.
remote: Compressing objects: 100% (152/152), done.
remote: Total 405976 (delta 173), reused 188 (delta 118), pack-reused 405706
Receiving objects: 100% (405976/405976), 120.38 MiB | 13.06 MiB/s, done.
Resolving deltas: 100% (269821/269821), done.
Submodule path 'buildroot': checked out '6b8e73d6a1958fcf441d91338c64c73aadcc43e4'

Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation
$ cd MycroftOS/

Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation/MycroftOS (develop)
$ git checkout develop
Already on 'develop'
Your branch is up to date with 'origin/develop'.

Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation/MycroftOS (develop)
$ git submodule update --remote --merge
Updating 6b8e73d6a1..efca26a405
Fast-forward
 CHANGES                                            |  37 +++++
 Makefile                                           |   4 +-
 linux/linux.hash                                   |   4 +-
 .../binutils/{2.35.1 => 2.35.2}/0001-sh-conf.patch |   0
 .../0002-poison-system-directories.patch           |   0
 package/binutils/Config.in.host                    |   4 +-
 package/binutils/binutils.hash                     |   2 +-
 package/busybox/udhcpc.script                      |   2 +-
 package/dhcpcd/dhcpcd.mk                           |   1 +
 ...d-fix-Werror=format-overflow-under-gcc-10.patch | 162 +++++++++++++++++++++
 package/intel-mediadriver/intel-mediadriver.mk     |   6 +-
 package/linux-headers/Config.in.host               |   4 +-
 package/mutt/0002-CVE-2021-3181-1.patch            |  40 +++++
 package/mutt/0003-CVE-2021-3181-2.patch            |  53 +++++++
 package/mutt/0004-CVE-2021-3181-3.patch            |  30 ++++
 package/mutt/mutt.mk                               |   5 +
 package/privoxy/privoxy.hash                       |   8 +-
 package/privoxy/privoxy.mk                         |   2 +-
 package/pugixml/pugixml.hash                       |   4 +-
 package/pugixml/pugixml.mk                         |   4 +-
 package/python-django/python-django.hash           |   6 +-
 package/python-django/python-django.mk             |   4 +-
 package/socat/socat.hash                           |  10 +-
 package/socat/socat.mk                             |   2 +-
 ...arch-sh-Add-64-bit-time-types-from-kernel.patch |  33 +++++
 .../wireguard-linux-compat.hash                    |   4 +-
 .../wireguard-linux-compat.mk                      |   2 +-
 .../xserver_xorg-server/xserver_xorg-server.mk     |   3 +
 package/xenomai/Config.in                          |   6 +-
 support/dependencies/check-host-cmake.mk           |   5 +
 support/misc/Vagrantfile                           |   2 +-
 31 files changed, 409 insertions(+), 40 deletions(-)
 rename package/binutils/{2.35.1 => 2.35.2}/0001-sh-conf.patch (100%)
 rename package/binutils/{2.35.1 => 2.35.2}/0002-poison-system-directories.patch (100%)
 create mode 100644 package/intel-mediadriver/0002-build-fix-Werror=format-overflow-under-gcc-10.patch
[... 213 lines cut ...]
config rpi3_64-gui
C:/Program Files (x86)/GnuWin32/bin/make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi3_64-gui_defconfig"
/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `C:/Program Files (x86)/GnuWin32/bin/make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi3_64-gui_defconfig"'
make: *** [rpi3_64-gui-config] Error 1

Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation/MycroftOS (develop)
$ make all
config rpi3_64-gui
C:/Program Files (x86)/GnuWin32/bin/make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi3_64-gui_defconfig"
/usr/bin/sh: -c: line 0: syntax error near unexpected token `('
/usr/bin/sh: -c: line 0: `C:/Program Files (x86)/GnuWin32/bin/make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi3_64-gui_defconfig"'
make: *** [rpi3_64-gui-config] Error 1

Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation/MycroftOS (develop)
$ make rpi3-config
make: *** No rule to make target `rpi3-config'.  Stop.

Logan@DESKTOP-ANBSM94 MINGW64 /c/Home automation/MycroftOS (develop)
$ make rpi3
make: *** No rule to make target `rpi3'.  Stop.

```

As you can see, fails in the make step, and I can't figure out why... any pointers?

ChanceNCounter, 2021-02-04:
I'm inclined to suspect MinGW. It's certainly betrayed me in the past. I could be wrong, being unfamiliar with the build process, but I can't even find that open paren, so I'm guessing it's a binary's fault.

If you don't have access to a Linux system, I'd suggest trying it in WSL. WSL has betrayed me far less frequently.

EVMakers, 2021-02-04:
https://github.com/OpenVoiceOS/OpenVoiceOS/issues/45

perhaps take a look here and try this device instead: **make rpi4_64-gui**
Pi 4 is the current development device from my understanding.

j1nx, 2021-02-04:
Heuhm, guess I need to update the documentation a bit and remove the no longer used config files although I don't know if MinGW will work. Guess we will know sooner then later.

TheLogan, 2021-02-04:
I discovered that there was also an issue of the path containing a space character, so I renamed "Home automation" to "HomeAutomation", deleted all files and started over, all with WSL, once I ran "make rpi4_64-gui I got this new error

```
root@DESKTOP-ANBSM94:/mnt/c/HomeAutomation/MycroftOS# make rpi4_64-gui
config rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi4_64-gui_defconfig"
make[1]: Entering directory '/mnt/c/HomeAutomation/MycroftOS/buildroot'
/bin/bash: gcc: command not found
/bin/bash: line 0: [: : integer expression expected
mkdir -p /mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="gcc" HOSTCC="gcc" \
    obj=/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config -C support/kconfig -f Makefile.br conf
make[2]: Entering directory '/mnt/c/HomeAutomation/MycroftOS/buildroot/support/kconfig'
gcc -DCURSES_LOC="<curses.h>" -DLOCALE -DKBUILD_NO_NLS -I/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/.depend 2>/dev/null || :
gcc -DCURSES_LOC="<curses.h>" -DLOCALE -DKBUILD_NO_NLS -I/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/conf.o
/bin/sh: 1: gcc: not found
Makefile.br:34: recipe for target '/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/conf.o' failed
make[2]: *** [/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/conf.o] Error 127
make[2]: Leaving directory '/mnt/c/HomeAutomation/MycroftOS/buildroot/support/kconfig'
Makefile:968: recipe for target '/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/conf' failed
make[1]: *** [/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/buildroot-config/conf] Error 2
make[1]: Leaving directory '/mnt/c/HomeAutomation/MycroftOS/buildroot'
Makefile:20: recipe for target 'rpi4_64-gui-config' failed
make: *** [rpi4_64-gui-config] Error 2
```

So I ran `apt install gcc`, it gave me a bunch of 404 errors, discovered that I needed to update my libraries or some such, ran `sudo apt update`, then installed gcc and finally ran make again, this time I'm being told I need to install g++, unzip and bc, so I'll go ahead and do that now xD
installed gcc, and then finally ran

TheLogan, 2021-02-04:
It ran for a long while this time but it ended with an error, and it seems to be this line:
 `mv: cannot stat '/mnt/c/HomeAutomation/MycroftOS/buildroot/output/build/glibc-2.31-74-gd0c84d22b6a67f85a1eed3b93aef30e6953294b5/build/dlfcn/stamp.oST': No such file or directory`
 
 The log is too large to post here xD

TheLogan, 2021-04-16:
Any eta on the documentation? (or a new release)

j1nx, 2021-05-05:
The README has been updated. Can you check if you start from scratch if it guides you through it?

TheLogan, 2021-05-05:
I'll try it after work :)

TheLogan, 2021-05-08:
Dangit, I'd written up the results of my test (it didn't work, in fact all 3 of the make commands failed), but I must have forgotten to click the "comment" button cause I can't seem to find the comment on this thread of any of the other ones.
I'll redo the run tomorrow when I get back to my computer!

TheLogan, 2021-05-09:
So this is `make clean`:
```
root@DESKTOP-ANBSM94:/mnt/c/HomeAutomation/OpenVoiceOS# make clean
make -C buildroot BR2_EXTERNAL=../buildroot-external clean
make[1]: Entering directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: 11: /mnt/c/HomeAutomation/OpenVoiceOS/build: not foundt/scripts/setlocalversion:
/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: 14: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: Syntax error: Bad fd number
: invalid optionbr2-external: line 2: set: -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
support/scripts/br2-external: line 3: $'\r': command not found
support/scripts/br2-external: line 7: $'\r': command not found
': not a valid identifiernal: line 9: declare: `BR2_EXT_NAMES
support/scripts/br2-external: line 10: $'\r': command not found
support/scripts/br2-external: line 14: $'\r': command not found
support/scripts/br2-external: line 15: syntax error near unexpected token `$'{\r''
'upport/scripts/br2-external: line 15: `main() {
Makefile:192: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/.br2-external.mk: No such file or directory
rm -rf /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/target /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/images /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/host  \
        /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/staging \
        /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/legal-info /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/graphs /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/per-package
make[1]: Leaving directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
```

This is `make rpi4_64-gui-config`
```
root@DESKTOP-ANBSM94:/mnt/c/HomeAutomation/OpenVoiceOS# make rpi4_64-gui-config
config rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi4_64-gui_defconfig"
make[1]: Entering directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: 11: /mnt/c/HomeAutomation/OpenVoiceOS/build: not foundt/scripts/setlocalversion:
/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: 14: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: Syntax error: Bad fd number
: invalid optionbr2-external: line 2: set: -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
support/scripts/br2-external: line 3: $'\r': command not found
support/scripts/br2-external: line 7: $'\r': command not found
': not a valid identifiernal: line 9: declare: `BR2_EXT_NAMES
support/scripts/br2-external: line 10: $'\r': command not found
support/scripts/br2-external: line 14: $'\r': command not found
support/scripts/br2-external: line 15: syntax error near unexpected token `$'{\r''
'upport/scripts/br2-external: line 15: `main() {
Makefile:192: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/.br2-external.mk: No such file or directory
make[1]: *** No rule to make target 'rpi4_64-gui_defconfig'.  Stop.
make[1]: Leaving directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
Makefile:20: recipe for target 'rpi4_64-gui-config' failed
make: *** [rpi4_64-gui-config] Error 2
```

And finally this is `make rpi4_64-gui`
```
root@DESKTOP-ANBSM94:/mnt/c/HomeAutomation/OpenVoiceOS# make rpi4_64-gui
config rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi4_64-gui_defconfig"
make[1]: Entering directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: 11: /mnt/c/HomeAutomation/OpenVoiceOS/build: not foundt/scripts/setlocalversion:
/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: 14: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/scripts/setlocalversion: Syntax error: Bad fd number
: invalid optionbr2-external: line 2: set: -
set: usage: set [-abefhkmnptuvxBCHP] [-o option-name] [--] [arg ...]
support/scripts/br2-external: line 3: $'\r': command not found
support/scripts/br2-external: line 7: $'\r': command not found
': not a valid identifiernal: line 9: declare: `BR2_EXT_NAMES
support/scripts/br2-external: line 10: $'\r': command not found
support/scripts/br2-external: line 14: $'\r': command not found
support/scripts/br2-external: line 15: syntax error near unexpected token `$'{\r''
'upport/scripts/br2-external: line 15: `main() {
Makefile:192: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/.br2-external.mk: No such file or directory
make[1]: *** No rule to make target 'rpi4_64-gui_defconfig'.  Stop.
make[1]: Leaving directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
Makefile:20: recipe for target 'rpi4_64-gui-config' failed
make: *** [rpi4_64-gui-config] Error 2
```

j1nx, 2021-05-09:
All looks like linux / windows and of line character non-sense, so guess we can confirm WSL is not supported. If you do not have access to a physical linux machine, I recommend to setup a virtual machine.

TheLogan, 2021-05-12:
I googled around a bit and discovered a solution. It turns out that
1) I was using WSL 1, there's a version 2 now, and 
2) that there's a settings file I had to create here: /etc/wsl.conf
I added the following to it
```
[interop]
enabled=false # enable launch of windows binaries; default is true
appendWindowsPath=false # append windows path to $PATH variable; default is true
```
And then went through the process again, this time it didn't change the line endings :)
I don't think it succeeded as my release as the final make command ends with 1 error, but I'd say I am a lot further this time around :)

I've appended the entirety of the output below:

```
root@DESKTOP-ANBSM94:/mnt/c/HomeAutomation/OpenVoiceOS# make clean
make -C buildroot BR2_EXTERNAL=../buildroot-external clean
make[1]: Entering directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
make[1]: Warning: File '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/.br2-external.mk' has modification time 0.29 s in the future
rm -rf /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/target /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/images /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/host  \
        /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/staging \
        /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/legal-info /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/graphs /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/per-package
make[1]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: Leaving directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
root@DESKTOP-ANBSM94:/mnt/c/HomeAutomation/OpenVoiceOS# make rpi4_64-gui-config
config rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi4_64-gui_defconfig"
make[1]: Entering directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
make[1]: Warning: File '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/.br2-external.mk' has modification time 0.29 s in the future
mkdir -p /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
    obj=/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config -C support/kconfig -f Makefile.br conf
make[2]: Entering directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/support/kconfig'
/usr/bin/gcc -DCURSES_LOC="<curses.h>" -DLOCALE  -I/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -MM *.c > /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/.depend 2>/dev/null || :
make[2]: Warning: File '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/.depend' has modification time 0.38 s in the future
/usr/bin/gcc -DCURSES_LOC="<curses.h>" -DLOCALE  -I/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config -DCONFIG_=\"\"   -c conf.c -o /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/conf.o
/usr/bin/gcc -DCURSES_LOC="<curses.h>" -DLOCALE  -I/mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config -DCONFIG_=\"\"  -I. -c /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/zconf.tab.c -o /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/zconf.tab.o
In file included from /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/zconf.tab.c:2484:
./util.c: In function ‘file_write_dep’:
./util.c:54:18: warning: ‘..config.tmp’ directive writing 12 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
   54 |  sprintf(buf, "%s..config.tmp", dir);
      |                  ^~~~~~~~~~~~
./util.c:54:2: note: ‘sprintf’ output between 13 and 4109 bytes into a destination of size 4097
   54 |  sprintf(buf, "%s..config.tmp", dir);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/buildroot-config/zconf.tab.c:2485:
./confdata.c: In function ‘conf_write’:
./confdata.c:778:23: warning: ‘.tmpconfig.’ directive writing 11 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
  778 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());
      |                       ^~~~~~~~~~~
./confdata.c:778:3: note: ‘sprintf’ output between 13 and 4119 bytes into a destination of size 4097
  778 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./confdata.c: In function ‘conf_write_autoconf’:
./confdata.c:989:18: warning: ‘.config.cmd’ directive writing 11 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
  989 |  sprintf(buf, "%s.config.cmd", dir);
      |                  ^~~~~~~~~~~
./confdata.c:989:2: note: ‘sprintf’ output between 12 and 4108 bytes into a destination of size 4097
  989 |  sprintf(buf, "%s.config.cmd", dir);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./confdata.c:995:18: warning: ‘.tmpconfig’ directive writing 10 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
  995 |  sprintf(buf, "%s.tmpconfig", dir);
      |                  ^~~~~~~~~~
./confdata.c:995:2: note: ‘sprintf’ output between 11 and 4107 bytes into a destination of size 4097
  995 |  sprintf(buf, "%s.tmpconfig", dir);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make[1]: *** [package/pkg-generic.mk:240: /mnt/c/HomeAutomation/OpenVoiceOS/buildroot/output/build/host-tar-1.32/.stamp_configured] Error 1
make[1]: Leaving directory '/mnt/c/HomeAutomation/OpenVoiceOS/buildroot'
cp -f buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
cp: cannot stat 'buildroot/output/images/sdcard.img': No such file or directory
make: *** [Makefile:26: rpi4_64-gui] Error 1
```

ChanceNCounter, 2021-07-10:
Now that we have Discussions, I'm gonna convert this to a thread, and name it to reflect that it *seems* to be an "on Windows" situation.

fwiw, this is heavier than anything I've ever done in WSL. Still, might be worth ongoing attempts.

closed 2021-07-10 by ChanceNCounter