OpenVoiceOS/ovos-buildroot #47: patch: hardcoded speech + bad enclosure assumptions

JarbasAl, 2021-02-18:
- removes hardcoded onboarding (speech + wifi + pairing message + mic muting....)
- moves readiness check to appropriate place

https://github.com/MycroftAI/mycroft-core/compare/dev...OpenVoiceOS:refactor/enclosure_assumptions

ChanceNCounter, 2021-02-18:
Is the TODO on `enclosure.base.py:74` blocking?

ChanceNCounter, 2021-02-18:
Hardcoded timeout on 133 may be objectionable.

AIIX, 2021-02-18:
Testing the patch still reports the "mycroft.skills.trained" message on bus to early even on first boot

ChanceNCounter, 2021-02-18:
Probably good to log (or at least print) status during that while loop, as it's in a core component and could cause Mycroft to (effectively) hang quietly.

JarbasAl, 2021-02-18:
https://github.com/MycroftAI/mycroft-core/compare/dev...OpenVoiceOS:refactor/enclosure_assumptions

should account for pairing now before reporting ready

JarbasAl, 2021-02-18:
> Hardcoded timeout on 133 may be objectionable.

that TODO comes from mycroft, its not blocking and is in general a safe assumption, anyway @j1nx can enforce that skills only boots after enclosure i think

j1nx, 2021-02-18:
Not really sure about that, but will investigate.

j1nx, 2021-05-05:
Systemd / Notifyd is implemented. Closing this one as the enclosure assumption patch is no longer valid with mycroft-lib.

closed 2021-05-05 by j1nx