OpenVoiceOS/ovos-buildroot #58: Build errors

limitless-code, 2021-08-01:
Build Environment

- Raspberry pi 4 2 GB
- Ubuntu 20.04.2 LTS

- gcc, gcc++
- subversion
- qttools5-dev
- qttools5-dev-tools
- python
- git clone latest copy of OpenVoiceOS as per README

warning occurs when running ```make rpi4_64-gui-config```: -

```
config rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi4_64-gui_defconfig"
make[1]: Entering directory '/home/ubuntu/OpenVoiceOS/buildroot'

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PROTOBUF
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=y]) && (BR2_HOSTARCH [=aarch64]=x86_64 || BR2_HOSTARCH [=aarch64]=x86)
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_PYCHROMECAST [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PROTOBUF
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=y]) && (BR2_HOSTARCH [=aarch64]=x86_64 || BR2_HOSTARCH [=aarch64]=x86)
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_PYCHROMECAST [=y]
#
# configuration written to /home/ubuntu/OpenVoiceOS/buildroot/.config
#
make[1]: Leaving directory '/home/ubuntu/OpenVoiceOS/buildroot'
```

Then after running ```make rpi4_64-gui ``` the following error occurs: -

```

/home/ubuntu/OpenVoiceOS/buildroot/output/host/bin/ccache /usr/bin/g++ -fno-PIE -c   -O2 -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o caller-save.o -MT caller-save.o -MMD -MP -MF ./.deps/caller-save.TPo ../../gcc/caller-save.c
/home/ubuntu/OpenVoiceOS/buildroot/output/host/bin/ccache /usr/bin/g++ -fno-PIE -c   -O2 -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include -I/home/ubuntu/OpenVoiceOS/buildroot/output/host/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o calls.o -MT calls.o -MMD -MP -MF ./.deps/calls.TPo ../../gcc/calls.c
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[4]: *** [Makefile:1117: insn-emit.o] Error 1
make[4]: *** Waiting for unfinished jobs....
rm gcc.pod
make[3]: *** [Makefile:4376: all-gcc] Error 2
make[2]: *** [package/pkg-generic.mk:250: /home/ubuntu/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/.stamp_built] Error 2
make[1]: *** [Makefile:84: _all] Error 2
make[1]: Leaving directory '/home/ubuntu/OpenVoiceOS/buildroot'
```
Not sure what dependency i am missing re: warning? Or if I am doing anything obviously wrong?

Thank you!

ChanceNCounter, 2021-08-01:
Where'd you find a link? I need to go tell people not to build this for a while.

ChanceNCounter, 2021-08-01:
Slightly longer version: wait a few days and we'll provide an image.

limitless-code, 2021-08-01:
> 
> 
> Where'd you find a link? I need to go tell people not to build this for a while.

OK thanks, I will hold fire.

I followed the instructions in the README of this repo. This link git clone --recurse-submodules https://github.com/OpenVoiceOS/OpenVoiceOS.git.

ChanceNCounter, 2021-08-01:
Oh, absolutely. I meant, who linked you to the repo? =P I thought the main place people were finding us was the corresponding thread at the Mycroft forum, where we periodically mention that it's Almost There. But if there's someplace else, where we've left things looking like it should work, I definitely need to go warn those people!

ChanceNCounter, 2021-08-01:
Now that I've pinned an issue for everyone else, I can take a bit more time to expand in here ;)

The gist is, the image is building nicely for the 2/5 of our developers who build it. Off the top of my head, I can't remember what distro/s they're running, but I'll ask.

Getting it to run on other distros means, as you discovered, going through dep hell on each of those systems. It's (usually) a one-time problem, apart from individual breakages on individual system updates down the road. But it's a time-consuming problem, and, at this prealpha moment, the manifest is subject to sudden change.

Hence, we might spend a few hours fighting through a chunk of dep hell that won't exist tomorrow, or we might tack on another whole dep tree by this time next week. It's not that we don't *wanna* help you guys, it's that *holy shit* could we (and you) end up spending extra time on Schroedinger's Dephell.

limitless-code, 2021-08-01:
Many thanks for taking the time to explain. I found myself here through curiosity after seeing OpenVoiceOS/MycroftOS mentioned on a few YouTube videos. I did a search for OpenVoiceOS (not google but Qwant) and that resulted in me ending up here. I understand the dephell issue so no pressure from me.  Looks like a fantastic project even in preAlpha from the YouTube videos I have seen. I look forward to testing any releases.

ChanceNCounter, 2021-08-01:
Ah! I forgot about the shout-out, didn't think about search engines. Makes sense.

For what it's worth, currently in numbered pre/alpha are [OVOS Skill Manager](https://github.com/OpenVoiceOS/ovos_skill_manager) and several of the skills in our Skills Store (available via OSM). OSM is standalone, and should work on any Mycroft build or device (testers sought!) and does not conflict with its first-party counterpart.

It offers the same functionality as MSM, the skill manager that comes with Mycroft, but it also provides access to several repositories that aren't supported by MycroftAI, Inc. (nor, apart from the OVOS Skill Store, by us.)

This includes the repos for other Mycroft-based projects, Pling, and a use-at-your-even-more-than-usual-peril interface for Andlo's Skill List, which the eponymous community member produced by crawling GitHub for everything that looks like a Mycroft skill.

ccoreilly, 2021-08-03:
I could successfully build OVOS in gitlab CI using a [builder docker image](https://github.com/ccoreilly/OpenVoiceOS/blob/catala/builder.Dockerfile). You can try building it locally from within the docker container, it contains all necessary dependencies.

NeonDaniel, 2021-08-05:
> > Where'd you find a link? I need to go tell people not to build this for a while.
> 
> OK thanks, I will hold fire.
> 
> I followed the instructions in the README of this repo. This link git clone --recurse-submodules https://github.com/OpenVoiceOS/OpenVoiceOS.git.

If you still want to do this and don't mind a little troubleshooting, my suspicion is a system dependency j1nx and I forgot to document because we already had it installed.. You might try installing the Mycroft apt dependencies and re-running.

https://github.com/MycroftAI/mycroft-core/blob/170ebc44ecf5cf3ceeaf44f282c992754b1bbc1b/dev_setup.sh#L342-L346

Also, want to make sure you ran `./scripts/br-patches.sh` per the setup instructions.

limitless-code, 2021-08-06:
Many thanks for the suggestions. Will try this over the weekend as I am still getting the same error with the docker container suggestion using builder.Dockerfile. All suggestions are appreciated regardless of outcome. :)

https://github.com/MycroftAI/mycroft-core/blob/170ebc44ecf5cf3ceeaf44f282c992754b1bbc1b/dev_setup.sh#L342-L346

limitless-code, 2021-08-07:
_oh my giddy I/O!_ finally got past the first dependency warning (not managed to compile it fully so can't test) and found the reason for the other error.

```
/home/ovos/OpenVoiceOS/buildroot/output/host/bin/ccache /usr/bin/g++ -fno-PIE -c   -O2 -I/home/ovos/OpenVoiceOS/buildroot/output/host/include -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include -I/home/ovos/OpenVoiceOS/buildroot/output/host/include -I/home/ovos/OpenVoiceOS/buildroot/output/host/include -I/home/ovos/OpenVoiceOS/buildroot/output/host/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o bb-reorder.o -MT bb-reorder.o -MMD -MP -MF ./.deps/bb-reorder.TPo ../../gcc/bb-reorder.c
`g++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
Makefile:1117: recipe for target 'insn-emit.o' failed
make[3]: *** [insn-emit.o] Error 4
make[3]: *** Waiting for unfinished jobs....
rm gcc.pod
make[3]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/build/gcc'
Makefile:4376: recipe for target 'all-gcc' failed
make[2]: *** [all-gcc] Error 2
make[2]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/build'
package/pkg-generic.mk:247: recipe for target '/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/.stamp_built' failed
make[1]: *** [/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/.stamp_built] Error 2
make[1]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot'
cp -f buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
cp: cannot stat 'buildroot/output/images/sdcard.img': No such file or directory
Makefile:24: recipe for target 'rpi4_64-gui' failed
make: *** [rpi4_64-gui] Error 1
```

**in dmesg I find what looks to be the cause the problem!**

```
[ 9065.579559] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=cc1plus,pid=262417,uid=1001
[ 9065.579601] Out of memory: Killed process 262417 (cc1plus) total-vm:588296kB, anon-rss:531756kB, file-rss:0kB, shmem-rss:0kB, UID:1001 pgtables:1152kB oom_score_adj:0
[ 9065.668545] oom_reaper: reaped process 262417 (cc1plus), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

any ideas as to how much memory I should have to compile/ make?

as i said I am running the compilation/ make on a **_Raspberry pi 4 2 GB_**. I can't upgrade the memory on this, is there a way to limit what "cc1plus" uses so it doesn't hang the machine? or increasing swap size may help? or some other issue causing it?
Not sure if anyone has running the compilation on an RPI 4 before?
 
Thank you.

NeonDaniel, 2021-08-09:
> _oh my giddy I/O!_ finally got past the first dependency warning (not managed to compile it fully so can't test) and found the reason for the other error.
> 
> ```
> /home/ovos/OpenVoiceOS/buildroot/output/host/bin/ccache /usr/bin/g++ -fno-PIE -c   -O2 -I/home/ovos/OpenVoiceOS/buildroot/output/host/include -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include -I/home/ovos/OpenVoiceOS/buildroot/output/host/include -I/home/ovos/OpenVoiceOS/buildroot/output/host/include -I/home/ovos/OpenVoiceOS/buildroot/output/host/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/dpd -I../libdecnumber -I../../gcc/../libbacktrace   -o bb-reorder.o -MT bb-reorder.o -MMD -MP -MF ./.deps/bb-reorder.TPo ../../gcc/bb-reorder.c
> `g++: internal compiler error: Killed (program cc1plus)
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
> Makefile:1117: recipe for target 'insn-emit.o' failed
> make[3]: *** [insn-emit.o] Error 4
> make[3]: *** Waiting for unfinished jobs....
> rm gcc.pod
> make[3]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/build/gcc'
> Makefile:4376: recipe for target 'all-gcc' failed
> make[2]: *** [all-gcc] Error 2
> make[2]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/build'
> package/pkg-generic.mk:247: recipe for target '/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/.stamp_built' failed
> make[1]: *** [/home/ovos/OpenVoiceOS/buildroot/output/build/host-gcc-initial-10.2.0/.stamp_built] Error 2
> make[1]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot'
> cp -f buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
> cp: cannot stat 'buildroot/output/images/sdcard.img': No such file or directory
> Makefile:24: recipe for target 'rpi4_64-gui' failed
> make: *** [rpi4_64-gui] Error 1
> ```
> 
> **in dmesg I find what looks to be the cause the problem!**
> 
> ```
> [ 9065.579559] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=cc1plus,pid=262417,uid=1001
> [ 9065.579601] Out of memory: Killed process 262417 (cc1plus) total-vm:588296kB, anon-rss:531756kB, file-rss:0kB, shmem-rss:0kB, UID:1001 pgtables:1152kB oom_score_adj:0
> [ 9065.668545] oom_reaper: reaped process 262417 (cc1plus), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
> ```
> 
> any ideas as to how much memory I should have to compile/ make?
> 
> as i said I am running the compilation/ make on a **_Raspberry pi 4 2 GB_**. I can't upgrade the memory on this, is there a way to limit what "cc1plus" uses so it doesn't hang the machine?
> 
> **EDIT:** I think something like `make -j 4` or ```make -j `nproc` may work. Currently trying and will update here.
> 
> or increasing swap size may help? or some other issue causing it in the make process?
> Not sure if anyone has running the compilation on an RPI 4 before?
> 
> Thank you.

Not sure about running on a Pi; increasing swap might fix OOM errors, though it will take several days to build (for reference I think it took about 2 days in a VM with 4 Zen2 cores and 16GB RAM allocated. Also, make sure you have about 70GB available disk space where you run this (68GB used as of 4/28/2021). (Not sure if this all made it into docs)

limitless-code, 2021-08-09:
Thanks. That's the info I was after, I have made some progress regarding some of my issues. The OOM errors are most probably a red herring as i ditched my rp4 2GB version for my rpi4 4Gb version but usage hardly went over 1GB RAM and found that my docker container which I am running in interactive "bash" mode to troubleshoot is running on an attached USB SSD not the SD card(rpi4 4GB version). This is now seeming performing fine. Conclusion: running the build process on an SD card (rpi4 2GB version) caused the jobs(etc) to queue up in physical memory as SD cards are too slow to write/read to and eventually caused the host kernel to sacrifice the docker container I was using to build the image (just a theory!).

That being said and getting further. 

After running **_make rpi4_64-gui-config_** I am still getting: 

```
WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PROTOBUF
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=y]) && (BR2_HOSTARCH [=aarch64]=x86_64 || BR2_HOSTARCH [=aarch64]=x86)
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_PYCHROMECAST [=y]

WARNING: unmet direct dependencies detected for BR2_PACKAGE_PYTHON_PROTOBUF
  Depends on [n]: (BR2_PACKAGE_PYTHON [=n] || BR2_PACKAGE_PYTHON3 [=y]) && (BR2_HOSTARCH [=aarch64]=x86_64 || BR2_HOSTARCH [=aarch64]=x86)
  Selected by [y]:
  - BR2_PACKAGE_PYTHON_PYCHROMECAST [=y]
```

I cannot seem to fix this no matter what python packages I install including the suggested packages from dev_setup.sh

I tried just ignoring this warning and ran make _**rpi4_64-gui**_

but after ~7 hrs it runs through some python tests and fails 2 tests, the codec test and another one (I'll edit this comment with more detail after it has finished another build attempt).

then it exits ~1hr later with the following:

```
/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18/Modules/_ctypes/libffi/src/arm/sysv.S:410: Error: undefined symbol FFI_TYPE_STRUCT_VFP_FLOAT
 used as an immediate value
/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18/Modules/_ctypes/libffi/src/arm/sysv.S:411: Error: undefined symbol FFI_TYPE_STRUCT_VFP_DOUBL
E used as an immediate value
/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18/Modules/_ctypes/libffi/src/arm/sysv.S:450: Error: undefined symbol FFI_TYPE_STRUCT_VFP_FLOAT
 used as an immediate value
/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18/Modules/_ctypes/libffi/src/arm/sysv.S:453: Error: undefined symbol FFI_TYPE_STRUCT_VFP_DOUBL
E used as an immediate value

Python build finished, but the necessary bits to build these modules were not found:
_bsddb             _sqlite3           _tkinter
bsddb185           bz2                dbm
gdbm               readline           sunaudiodev
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_ctypes

Makefile:544: recipe for target 'sharedmods' failed
make[2]: *** [sharedmods] Error 1
make[2]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18'
package/pkg-generic.mk:247: recipe for target '/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18/.stamp_built' failed
make[1]: *** [/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18/.stamp_built] Error 2
make[1]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot'
```

Not sure how far into the build it is before exiting  (but it is getting much further than before) and not sure if this is a problem with packages not installed in my build environment or something else? Pretty sure my hardware issues are resolved now though and I have over 80GB available on my USB attached SSD so that should be fine.

NeonDaniel, 2021-08-10:
> make[2]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot/output/build/host-python-2.7.18'

Are you building with `python2.7`? 3.8 is what worked for me last; you can check with:
```
python --version
```

limitless-code, 2021-08-10:
You are correct I am. I will try the version 3.8

NeonDaniel, 2021-08-10:
> You are correct I am. I will try the version 3.8

If you're keeping track of fixes as you go, adding notes to the README would be much appreciated! :)

limitless-code, 2021-08-11:
will do.. however LOL:

I have built another docker image... I am sure it doesn't have python version 2.X but still the same warning and error for make rpi4_64-gui-config  and make rpi4_64-gui .

I am in my own python env this time too.

```
(my_env) ovos@c884047e7572:~/OpenVoiceOS$ python3
Python 3.8.10 (default, Jun  2 2021, 10:49:15)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

Note: Ubuntu latest is 

here is the Dockerfile

```
FROM ubuntu:latest

ENV TERM linux
ENV DEBIAN_FRONTEND noninteractive

RUN set -x \
    && sed -i 's/# \(.*multiverse$\)/\1/g' /etc/apt/sources.list \
    && apt-get update \
    && apt-get -y install --no-install-recommends \
    vim less nano git subversion gcc qttools5-dev qttools5-dev-tools python3 python3-dev libncurses-dev \
    patch make file g++ wget cpio unzip rsync bc git sudo ca-certificates xz-utils curl \
    qtbase5-dev qtdeclarative5-dev bzip2 gzip tar\
    python3-setuptools libtool \
    libffi-dev libssl-dev autoconf automake bison swig libglib2.0-dev \
    portaudio19-dev mpg123 screen flac curl libicu-dev pkg-config \
    libjpeg-dev libfann-dev build-essential jq pulseaudio pulseaudio-utils \
    && apt-get -y autoremove \
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*

RUN addgroup --gid 1001 --system ovos && \
    adduser --system --uid 1001 --gid 1001 ovos

USER ovos
```

I am running it in bash (entry point via command line) in the docker container the other thing I changed from the dockerfile was running this as root.

```
apt install python3.8-venv
```

could it be that buildroot/ make is downloading python-2.7.18 and is separate from my OS build environment?

limitless-code, 2021-08-11:
I am most likely wrong here as I don't understand how buildroot works or what the expected behaviour is.

however the modules it could not find are modules that a patch told it to disable so i guess this part where it can't find modules is expected behaviour.

e.g
 ```
Applying 0019-Add-option-to-disable-the-sqlite3-module.patch using patch:
```

so I guess it's this error:

```
Failed to build these modules:
_ctypes

Makefile:544: recipe for target 'sharedmods' failed
make[2]: *** [sharedmods] Error 1

```
that's pertinent to it failing, as it mentions python-2.7.18 during this process up to failing which is something that is separate to the OS build environment?

j1nx, 2021-08-11:
What build machine you are doing all this?

limitless-code, 2021-08-11:
1. Raspberry pi 4 
2. 4GB RAM
3. Raspbian buster
5. Running Docker
6. USB attached SSD for docker container

running docker container interactively with ubuntu 20.04 as per Dockerfile in above comments

```
sudo docker exec -it <container-name> bash
```

Then followed the README.

j1nx, 2021-08-11:
Right! Buildroot is a cross compiling system. You do not run it on a rpi but on a normal x86_64 system.

limitless-code, 2021-08-12:
Thank you for ending my hell! This makes total sense. I shall find a spare x86_64 system or use my laptop. I looked into emulating an x86_64 system in docker using qemu on the rpi 4 but decided not to, saving the little sanity I have left in the process as that still may not currently be possible on arm based architecture anyway (some emulation does work but not x86_64 currently, only manged to get i686 emulation working in docker)

So to conclude: **Only use x86_64 base architecture/OS**

NeonDaniel, 2021-08-12:
> So to conclude: **Only use x86_64 based architecture/OS**

If you could, please add a note to the README. Totally missed this in my troubleshooting.

limitless-code, 2021-08-12:
Sure, I will hold fire on conclusion 'til I receive new SSD so i can boot to it, test on x86_64 architecture and ensure nothing else is missed. 

That won't happen 'till the after weekend.

limitless-code, 2021-08-15:
new note: FTP port 21 needs to be allowed to the "internet"

limitless-code, 2021-08-17:
OK my build environment is: -

hardware: x86_64 architecture
OS: Ubuntu 20.04 desktop
Extra packages installed via apt in addition to the packages in the README:

1. git
2. make
3. g++
4. curl 

build errors out after ~15rs

```
>>> host-kf5-ki18n 5.75.0 Configuring
(mkdir -p /home/ovos/OpenVoiceOS/buildroot/output/build/host-kf5-ki18n-5.75.0//buildroot-build && cd /home/ovos/OpenVoiceOS/buildroot/output/build/host-kf5-ki18n-5.75.0//buildroot-build && rm -f CMakeCache.txt && PATH="/home/ovos/OpenVoiceOS/buildroot/output/host/bin:/home/ovos/OpenVoiceOS/buildroot/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PKG_CONFIG="/home/ovos/OpenVoiceOS/buildroot/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_LIBDIR="/home/ovos/OpenVoiceOS/buildroot/output/host/lib/pkgconfig:/home/ovos/OpenVoiceOS/buildroot/output/host/share/pkgconfig" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1  /home/ovos/OpenVoiceOS/buildroot/output/host/bin/cmake /home/ovos/OpenVoiceOS/buildroot/output/build/host-kf5-ki18n-5.75.0/ -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_FIND_ROOT_PATH="/home/ovos/OpenVoiceOS/buildroot/output/host" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM="BOTH" -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY="BOTH" -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE="BOTH" -DCMAKE_INSTALL_PREFIX="/home/ovos/OpenVoiceOS/buildroot/output/host" -DCMAKE_C_FLAGS="-O2 -I/home/ovos/OpenVoiceOS/buildroot/output/host/include" -DCMAKE_CXX_FLAGS="-O2 -I/home/ovos/OpenVoiceOS/buildroot/output/host/include" -DCMAKE_EXE_LINKER_FLAGS="-L/home/ovos/OpenVoiceOS/buildroot/output/host/lib -Wl,-rpath,/home/ovos/OpenVoiceOS/buildroot/output/host/lib" -DCMAKE_SHARED_LINKER_FLAGS="-L/home/ovos/OpenVoiceOS/buildroot/output/host/lib -Wl,-rpath,/home/ovos/OpenVoiceOS/buildroot/output/host/lib" -DCMAKE_ASM_COMPILER="/usr/bin/as" -DCMAKE_C_COMPILER="/usr/bin/gcc" -DCMAKE_CXX_COMPILER="/usr/bin/g++"  -DCMAKE_C_COMPILER_LAUNCHER="/home/ovos/OpenVoiceOS/buildroot/output/host/bin/ccache" -DCMAKE_CXX_COMPILER_LAUNCHER="/home/ovos/OpenVoiceOS/buildroot/output/host/bin/ccache"  -DCMAKE_COLOR_MAKEFILE=OFF -DBUILD_DOC=OFF -DBUILD_DOCS=OFF -DBUILD_EXAMPLE=OFF -DBUILD_EXAMPLES=OFF -DBUILD_TEST=OFF -DBUILD_TESTS=OFF -DBUILD_TESTING=OFF -DBUILD_SHARED_LIBS=ON  -DCMAKE_CXX_FLAGS="-O2 -I/home/ovos/OpenVoiceOS/buildroot/output/host/include" )
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 

Installing in /home/ovos/OpenVoiceOS/buildroot/output/host. Run /home/ovos/OpenVoiceOS/buildroot/output/build/host-kf5-ki18n-5.75.0/buildroot-build/prefix.sh to set the environment for KI18n.
-- Looking for __GLIBC__
-- Looking for __GLIBC__ - found
-- Performing Test _OFFT_IS_64BIT
-- Performing Test _OFFT_IS_64BIT - Success
-- Performing Test HAVE_DATE_TIME
-- Performing Test HAVE_DATE_TIME - Success
CMake Error at CMakeLists.txt:42 (find_package):
  Could not find a package configuration file provided by "Qt5Qml" (requested
  version 5.12.0) with any of the following names:

    Qt5QmlConfig.cmake
    qt5qml-config.cmake

  Add the installation prefix of "Qt5Qml" to CMAKE_PREFIX_PATH or set
  "Qt5Qml_DIR" to a directory containing one of the above files.  If "Qt5Qml"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/ovos/OpenVoiceOS/buildroot/output/build/host-kf5-ki18n-5.75.0/buildroot-build/CMakeFiles/CMakeOutput.log".
make[2]: *** [package/pkg-generic.mk:240: /home/ovos/OpenVoiceOS/buildroot/output/build/host-kf5-ki18n-5.75.0/.stamp_configured] Error 1
make[1]: *** [Makefile:84: _all] Error 2
make[1]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot'
cp -f buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
cp: cannot stat 'buildroot/output/images/sdcard.img': No such file or directory
make: *** [Makefile:26: rpi4_64-gui] Error 1
```

The build environment shows "libqt5qml5 is already the newest version (5.12.8-0ubuntu1)."

limitless-code, 2021-08-17:
sudo apt-get install qtdeclarative5-dev fixes the above error.

ChanceNCounter, 2021-08-18:
On it.

ChanceNCounter, 2021-08-18:
Re: moved packages, #76

limitless-code, 2021-08-25:
It seems that Mycroft loaded my some of skills before pairing? Only noticed this when I opened my front door home assistant told Mycroft (OVOS) and Mycroft spoke that my "front door is open", didn't think it should do this before pairing. It was stuck on the Mycroft/ Local Backend choice for a while and rebooted a few times so didn't get a chance to get the pairing code in order to pair.

Have a Home assistant skill setup in home.mycroft.ai from previous picroft installation, not sure how it knew to download "my profile" before pairing (mac address? just a guess as I'm pretty sure my public ip address has change since picroft install).

JarbasAl, 2021-08-25:
the local backend selection path is not yet fully implemented, it will seem to get stuck and never finish loading