OpenVoiceOS/ovos-buildroot #65: pre-alpha 08092005 mycroft backend issues

Olzeke51, 2021-08-09:
on a M2-devkit, with the pre-alpha, selected the mycroft backend
got device paired 
later wanted to load ssh key - BUT ALL the Geo Location data was missing
.    AND I couldn't save any of the re-entered data - mycroft through an   "error updating device"
.
will try the local backend and get the logs

j1nx, 2021-08-10:
On the current development image SSH is turned on by default with "normal" user/password system.

user = mycroft
Passwd = mycroft

just find out the ip address of the device and just connect to its ip via ssh and log in with above credentials.

unfortunately on the dev kit currentle without the mic you can't ask for its ip address so you need to look at your router or similar.

later official images will have SSH turned off by default, but then you will be able to turn in on by either voice, settings or GUI control.

JarbasAl, 2021-08-10:
lets be sure to add an option to skill settings to not allow enabling by voice/GUI, i might want to lock it down and its a moot point if anyone with physical access to the device can enable it

Olzeke51, 2021-08-10:
OPERATOR/USER error - I deleted the "identity2.json" file  -
SSH is a seperate design issue

closed 2021-08-10 by Olzeke51

ChanceNCounter, 2021-08-10:
> lets be sure to add an option to skill settings to not allow enabling by voice/GUI, i might want to lock it down and its a moot point if anyone with physical access to the device can enable it that easily
> 
> If it requires taking of the sd card/usb drive its much easier to detect an intrusion

Put it on a keyboard shortcut, so you can only do it by plugging in a physical keyboard. These are not viable options, however, for those who use voice devices because their hands don't work.

We could put a touchscreen PIN/voice password on certain such things, but then you'd have to do a factory reset if you forgot the PIN.

We could do a sort of 2FA with the backend, assuming one exists, but this doesn't help when none is present.