OpenVoiceOS/ovos-buildroot #66: pre-alpha 08092005 shutdown/restart not working

Olzeke51, 2021-08-09:
M2-devkit ,attempting to leave the mycroft-backend and restart unit (NO SSH available)
{I had selected 'enable SSH' from the menu - FWI no confirmation or anything}
.
selected both 'restart icon' and also the pwr-down icon but it only returns me to homescreen
expected a terminal screen with data - similar to mycroft
.
just turned off power

ChanceNCounter, 2021-08-09:
ssh should be working out of the box. i was able to ssh in as soon as it joined wifi.

Expect an update later today or tomorrow (US time) regarding which parts of the touchscreen UI are hooked up. I know the volume slider isn't. Sure is sexy, though!

j1nx, 2021-08-10:
We will soon look into connecting the different dots from the GUI menu. Reboot and Shutdown did work before so not sure what regression happened there but we will look into it.