OpenVoiceOS/ovos-buildroot #74: almost-alpha-08161630 fails to fully boot up

Olzeke51, 2021-08-16:
Mk2-devkit, with SJ and network cable, booting via USB flash
log files
* nnote - 2nd USB flash [both SanDisk 16G] also fails - used dd with bs=512K
* 
[08161630-img-failed-LOGS.tar.gz](https://github.com/OpenVoiceOS/OpenVoiceOS/files/6995697/08161630-img-failed-LOGS.tar.gz)

ChanceNCounter, 2021-08-16:
how far does it get?

Olzeke51, 2021-08-16:
09:39 it reboots after setting the backend & Kaldi
11:11 goes dark - no backlight
11:18 a flash of white on screen
11:27 'loading bar' pops up 
11:41 a Mycroft red ring of leds
11:59 screen is blanked black (has backlight)
12:06 ovos logo with upper right corner spinner -   forever

ChanceNCounter, 2021-08-16:
We'll have to narrow that down. I also got lousy performance over USB, but that could have been the device. I've had much better results on an SD card.

However, it could also be the combo of local backend and Kaldi, somehow. If you have time, I'd appreciate it if you tried changing one of those variables (use an SD card, or a different wake word, or Mycroft backend)

I'd try to repro myself, but I'm currently working on the only SD card I can spare for OVOS =P

Olzeke51, 2021-08-17:
just now saw this 
I did finally get a good load  - took a long time
can't seem to find any .conf files other than 'system'
mycroft-config show  user/default/remote  - no files [course some will fail as not paired to Mycroft]
*
in the earlier version of pre-alpha - had not much luck with home.mycroft.ai - it would pair
but couldn't change any of the skills/ lost device connection/ pairing info - - per the logs [my earlier 'issue']

haven't checked logs to see when / how long it was for a 'GOOD'  
btw- it says I am in Kansas  [via the cli-client]
[ovos-08161630-GOOD-logs.tar.gz](https://github.com/OpenVoiceOS/OpenVoiceOS/files/6996064/ovos-08161630-GOOD-logs.tar.gz)

Olzeke51, 2021-08-17:
per Chance's request - did a 'stop' from the menu
on pwr-cycle did a local / Google selection
overall it took close to 8 minutes to get to 'homescreen/standby face'

ChanceNCounter, 2021-08-17:
@j1nx is probably in order =P

Maybe it was expanding the filesystem? But that doesn't explain the forever-hang.

ChanceNCounter, 2021-08-17:
Investigation: able to repro off an SD card. I think the problem is Kaldi. Gotta try it a few more times and get logs at each step, but I'm giving it a rest for the moment. Steps:

* Onboarding: Local/Kaldi
* Wait for "forever spinner"
* ssh in and ensure Mycroft is running
* trigger STT, expect to be kicked back to backend selection

ChanceNCounter, 2021-08-17:
@AIIX I can't find any individual selection that changes the behavior. Is it expected to hang before kicking the user back?

Olzeke51, 2021-08-17:
"forever" might be like an hour and a half  - plugged my earlier unit back in
came back 1.5 hrs later and had standby face

Olzeke51, 2021-08-17:
it seems that the OOB initial run with local backend and STT==Kaldi may have a 'building' process going on for 1-2 hours
subsequent reboots take about 8 minutes

JarbasAl, 2021-08-17:
local backend code path is not yet completed, in current image you need to use selene

ChanceNCounter, 2021-08-17:
yes, but the behavior is more complicated, even if the diagnosis isn't. at present, rebooting (or dialing in and restarting services) will get you the home screen, and you can use simple intents via `mycroft-say-to`. Only when you activate STT, in my experience, does it kick you back to setup.

j1nx, 2023-12-29:
Should be fixed by now

closed 2023-12-29 by j1nx