OpenVoiceOS/ovos-buildroot #77: almost-alpha 0816 - funky response from mycroft-say-to re:time request

Olzeke51, 2021-08-18:
flashed new image to usb / mark2-devkit - network cable
selene/Mycroft backend selected
mycroft-say-to what's the time       **got a prompt instead of time ???????
'''mycroft@OpenVoiceOS-e37cd7:~ $ pacmd set-default-sink 0
mycroft@OpenVoiceOS-e37cd7:~ $ mycroft-speak this is really cool stuff
mycroft@OpenVoiceOS-e37cd7:~ $ mycroft-say-to sing for me
mycroft@OpenVoiceOS-e37cd7:~ $ mycroft-say-to sing to me
mycroft@OpenVoiceOS-e37cd7:~ $ mycroft-say-to what's the time
> ^C
mycroft@OpenVoiceOS-e37cd7:~ $ mycroft-say-to what is the time
mycroft@OpenVoiceOS-e37cd7:~ $ mycroft-say-to what's the time
> ls
> help
> list
> exit
> bye
> ^C
mycroft@OpenVoiceOS-e37cd7:~ $ 
'''
**short time span ~40 minutes altogether

[mycroft-backend-fresh.tar.gz](https://github.com/OpenVoiceOS/OpenVoiceOS/files/7009959/mycroft-backend-fresh.tar.gz)

Olzeke51, 2021-08-18:
this also happens to "what's the weather" vs 'what is the weather'

j1nx, 2023-12-29:
should be fixed, if not let me know

closed 2023-12-29 by j1nx