OpenVoiceOS/ovos-buildroot #86: Files missing when patching

limitless-code, 2021-09-04:
Version: Newest dev git clone.

```
./scripts/br-patches.sh 
can't find file to patch at input line 50
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 048dd3f7be74a2155d6742731efe46ee076f7b5d Mon Sep 17 00:00:00 2001
|From: j1nx <p.steenbergen@j1nx.nl>
|Date: Wed, 7 Apr 2021 17:01:51 +0200
|Subject: [PATCH 1/1] Switch Python3.9.4 to Python3.8.9
|
|---
| ...e-the-build-of-pyc-files-conditional.patch | 14 +++----
| ...taddrinfo-configure-test-when-cross-.patch |  8 ++--
| ...re-to-disable-the-build-of-certain-e.patch | 20 +++++-----
| ...y-header-paths-for-cross-compilation.patch | 12 +++---
| ...ook-in-usr-lib-termcap-for-libraries.patch |  8 ++--
| .../0006-Don-t-add-multiarch-paths.patch      |  8 ++--
| .../0007-Abort-on-failed-module-build.patch   |  8 ++--
| .../0008-Serial-ioctl-workaround.patch        |  6 +--
| ...e-shebang-of-Python-scripts-for-cros.patch |  4 +-
| ...ig.sh.in-ensure-sed-invocations-only.patch |  4 +-
| ...locale-and-set-to-default-when-addin.patch | 14 +++----
| ...Add-importlib-fix-for-PEP-3147-issue.patch | 12 +++---
| ...-disable-installation-of-test-module.patch | 29 ++++++--------
| .../0014-Add-an-option-to-disable-pydoc.patch | 28 ++++++-------
| ...015-Add-an-option-to-disable-lib2to3.patch | 24 +++++------
| ...option-to-disable-the-sqlite3-module.patch | 16 ++++----
| ...d-an-option-to-disable-the-tk-module.patch | 20 +++++-----
| ...-option-to-disable-the-curses-module.patch | 26 ++++++------
| .../0019-Add-an-option-to-disable-expat.patch | 20 +++++-----
| ...-Add-an-option-to-disable-CJK-codecs.patch |  8 ++--
| .../0021-Add-an-option-to-disable-NIS.patch   |  8 ++--
| ...Add-an-option-to-disable-unicodedata.patch |  8 ++--
| .../0023-Add-an-option-to-disable-IDLE.patch  | 20 +++++-----
| ...024-Add-an-option-to-disable-decimal.patch | 16 ++++----
| ...on-to-disable-the-ossaudiodev-module.patch |  8 ++--
| ...an-option-to-disable-openssl-support.patch |  8 ++--
| ...ption-to-disable-the-readline-module.patch |  8 ++--
| ...to-disable-zlib-bzip2-and-xz-modules.patch |  8 ++--
| ...thon-config.sh-don-t-reassign-prefix.patch |  4 +-
| ...-Fix-cross-compiling-the-uuid-module.patch |  8 ++--
| ...Add-an-option-to-disable-uuid-module.patch |  8 ++--
| ...-fix-building-on-older-distributions.patch |  4 +-
| ...up-CC-print-multiarch-output-for-mus.patch |  8 ++--
| ...-ng-doesn-t-set-errno-when-encryptio.patch | 40 -------------------
| package/python3/python3.hash                  |  5 +--
| package/python3/python3.mk                    |  7 +++-
| 36 files changed, 205 insertions(+), 252 deletions(-)
| delete mode 100644 package/python3/0034-lib-crypt-uClibc-ng-doesn-t-set-errno-when-encryptio.patch
|
|diff --git a/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch b/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
|index 5eb296c6d8..1010d08d53 100644
|--- a/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
|+++ b/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
5 out of 5 hunks ignored
can't find file to patch at input line 102
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/package/python3/0002-Disable-buggy_getaddrinfo-configure-test-when-cross-.patch b/package/python3/0002-Disable-buggy_getaddrinfo-configure-test-when-cross-.patch
|index c291d14460..e9e5b32ca2 100644
|--- a/package/python3/0002-Disable-buggy_getaddrinfo-configure-test-when-cross-.patch
|+++ b/package/python3/0002-Disable-buggy_getaddrinfo-configure-test-when-cross-.patch
[... 415 lines cut ...]
The text leading up to this was:
--------------------------
|diff --git a/package/python3/python3.hash b/package/python3/python3.hash
|index 2a0ba6ab4f..b7c3963fa6 100644
|--- a/package/python3/python3.hash
|+++ b/package/python3/python3.hash
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 1541
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/package/python3/python3.mk b/package/python3/python3.mk
|index 345c6a8699..0153a8e90f 100644
|--- a/package/python3/python3.mk
|+++ b/package/python3/python3.mk
--------------------------
File to patch: 
Skip this patch? [y] 
Skipping patch.
2 out of 2 hunks ignored
```

limitless-code, 2021-09-07:
Doh! not sure what I did there, thanks!

closed 2021-09-07 by limitless-code

limitless-code, 2021-09-08:
I suspect but cannot confirm as it's currently building and patched successfully, that it may have been because I did a fresh clone with ``` git clone https://github.com/OpenVoiceOS/OpenVoiceOS.git ```

instead of

``` git clone --recurse-submodules https://github.com/OpenVoiceOS/OpenVoiceOS.git ```

a fresh git clone using --recurse-submodules (which I had done up until this time, whoops) didn't give me the above patching errors.

j1nx, 2021-09-08:
Ah right, yes that explains it.

Without the --recursive-submodules you are not checking out the tagged buildroot release inside the buildroot folder, so that folder is empty so contains no files to patch.

limitless-code, 2021-09-08:
Yes that makes sense. Thanks for confirming.