OpenVoiceOS/ovos-buildroot #102: FindKF5DocTools error when running 'make rpi3_64-gui'

dickorydock, 2022-03-09:
I am building for the RPi3 on Ubuntu 20.04 and wanted to share an error I got in case it is helpful for you all developing this great software. It seems kind of similar to the libGLESv4 issue mentioned in the documentation but not the same. 
```

-- Looking for __GLIBC__
-- Looking for __GLIBC__ - found
-- Performing Test _OFFT_IS_64BIT
-- Performing Test _OFFT_IS_64BIT - Success
-- Performing Test HAVE_DATE_TIME
-- Performing Test HAVE_DATE_TIME - Success
CMake Warning at CMakeLists.txt:26 (find_package):
  By not providing "FindKF5DocTools.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "KF5DocTools", but CMake did not find one.

  Could not find a package configuration file provided by "KF5DocTools"
  (requested version 5.82.0) with any of the following names:

    KF5DocToolsConfig.cmake
    kf5doctools-config.cmake

  Add the installation prefix of "KF5DocTools" to CMAKE_PREFIX_PATH or set
  "KF5DocTools_DIR" to a directory containing one of the above files.  If
  "KF5DocTools" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at CMakeLists.txt:27 (find_package):
  By not providing "FindKF5Service.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "KF5Service", but CMake did not find one.

  Could not find a package configuration file provided by "KF5Service"
  (requested version 5.82.0) with any of the following names:

    KF5ServiceConfig.cmake
    kf5service-config.cmake

  Add the installation prefix of "KF5Service" to CMAKE_PREFIX_PATH or set
  "KF5Service_DIR" to a directory containing one of the above files.  If
  "KF5Service" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

```

j1nx, 2022-03-09:
Indeed! Ran into the same error, working on the x86_64 Open Virtual Appliance, however that work has not yet been pusjed or merged because it is wip.

For you to fix it, you can change;

diff --git a/buildroot-external/package/kf5-kded/kf5-kded.mk b/buildroot-external/package/kf5-kded/kf5-kded.mk
index c591c73..a375719 100644
--- a/buildroot-external/package/kf5-kded/kf5-kded.mk
+++ b/buildroot-external/package/kf5-kded/kf5-kded.mk
@@ -10,7 +10,7 @@ KF5_KDED_SOURCE = kded-$(KF5_KDED_VERSION).tar.xz
 KF5_KDED_LICENSE = BSD-3-Clause
 KF5_KDED_LICENSE_FILES = COPYING-CMAKE-SCRIPTS

-KF5_KDED_DEPENDENCIES = host-pkgconf
+KF5_KDED_DEPENDENCIES = host-pkgconf kf5-kservice
 KF5_KDED_INSTALL_STAGING = YES
 KF5_KDED_SUPPORTS_IN_SOURCE_BUILD = NO

Then rerun the "make rp3_64-gui" command.

j1nx, 2022-03-09:
Ah right, that error is still there as well. Crap.

Will post a quick fix to get past it a little bit later today.

j1nx, 2022-03-09:
Apologies, it will take a bit longer as I am trying to reproduce the error again.

dickorydock, 2022-03-09:
That's fine! Take your time. Let me know if there's other information I can provide in the meanwhile. Thanks

dickorydock, 2022-04-06:
Hello, checking if there are any updates on this error. Thank you.

RedB00t, 2022-06-13:
Found this thread due to having the same issue as above.... just posting to see if there is an update and thank you for all that you do!

NHellFire, 2023-04-14:
For the desktoptojson issue, you'll need to install the system package providing it (for Debian flavours, that's libkf5coreaddons-dev-bin). I'm currently looking at getting that built by buildroot instead.

j1nx, 2023-04-21:
> For the desktoptojson issue, you'll need to install the system package providing it (for Debian flavours, that's libkf5coreaddons-dev-bin). I'm currently looking at getting that built by buildroot instead. Same for kf5-kconfig (then #128 won't be needed).

I hope you figure out a smart way to do this indeed as buildroot host package, however buildroot does not have a QT5 host package so when I tried it ended up in a rabbithole.

j1nx, 2023-12-29:
no longer needed

closed 2023-12-29 by j1nx