OpenVoiceOS/ovos-buildroot #109: ./scripts/br-patches.sh fails to find file

vvstubbs, 2022-04-18:
When running patch script:

⌊~/git/OpenVoiceOS (develop) $ ./scripts/br-patches.sh
can't find file to patch at input line 52
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 6a994bf9f58563e9f4c920037532c9590ccb78ed Mon Sep 17 00:00:00 2001
|From: Peter Steenbergen <info@j1nx.nl>
|Date: Mon, 13 Sep 2021 08:24:10 +0200
|Subject: [PATCH 1/1] Switch Python3.9.7 to Python3.8.12
|
|---
| ...e-the-build-of-pyc-files-conditional.patch | 14 +++----
| ...taddrinfo-configure-test-when-cross-.patch |  8 ++--
| ...re-to-disable-the-build-of-certain-e.patch | 20 +++++-----
| ...y-header-paths-for-cross-compilation.patch | 12 +++---
| ...ook-in-usr-lib-termcap-for-libraries.patch |  8 ++--
| .../0006-Don-t-add-multiarch-paths.patch      |  8 ++--
| .../0007-Abort-on-failed-module-build.patch   |  8 ++--
| .../0008-Serial-ioctl-workaround.patch        |  6 +--
| ...e-shebang-of-Python-scripts-for-cros.patch |  4 +-
| ...ig.sh.in-ensure-sed-invocations-only.patch |  4 +-
| ...locale-and-set-to-default-when-addin.patch | 14 +++----
| ...Add-importlib-fix-for-PEP-3147-issue.patch | 12 +++---
| ...-disable-installation-of-test-module.patch | 29 ++++++--------
| .../0014-Add-an-option-to-disable-pydoc.patch | 28 ++++++-------
| ...015-Add-an-option-to-disable-lib2to3.patch | 24 +++++------
| ...option-to-disable-the-sqlite3-module.patch | 16 ++++----
| ...d-an-option-to-disable-the-tk-module.patch | 20 +++++-----
| ...-option-to-disable-the-curses-module.patch | 26 ++++++------
| .../0019-Add-an-option-to-disable-expat.patch | 20 +++++-----
| ...-Add-an-option-to-disable-CJK-codecs.patch |  8 ++--
| .../0021-Add-an-option-to-disable-NIS.patch   |  8 ++--
| ...Add-an-option-to-disable-unicodedata.patch |  8 ++--
| .../0023-Add-an-option-to-disable-IDLE.patch  | 20 +++++-----
| ...024-Add-an-option-to-disable-decimal.patch | 16 ++++----
| ...on-to-disable-the-ossaudiodev-module.patch |  8 ++--
| ...an-option-to-disable-openssl-support.patch |  8 ++--
| ...ption-to-disable-the-readline-module.patch |  8 ++--
| ...to-disable-zlib-bzip2-and-xz-modules.patch |  8 ++--
| ...thon-config.sh-don-t-reassign-prefix.patch |  4 +-
| ...-Fix-cross-compiling-the-uuid-module.patch |  8 ++--
| ...Add-an-option-to-disable-uuid-module.patch |  8 ++--
| ...-fix-building-on-older-distributions.patch |  4 +-
| ...up-CC-print-multiarch-output-for-mus.patch |  8 ++--
| ...-ng-doesn-t-set-errno-when-encryptio.patch | 40 -------------------
| ...teration-error-in-_ExecutorManagerTh.patch | 29 --------------
| package/python3/python3.hash                  |  5 +--
| package/python3/python3.mk                    |  4 +-
| 37 files changed, 202 insertions(+), 281 deletions(-)
| delete mode 100644 package/python3/0034-lib-crypt-uClibc-ng-doesn-t-set-errno-when-encryptio.patch
| delete mode 100644 package/python3/0036-Fix-dictionary-iteration-error-in-_ExecutorManagerTh.patch
|
|diff --git a/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch b/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
|index 44902725d7..1010d08d53 100644
|--- a/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
|+++ b/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
--------------------------
File to patch: 

happens every time, even tired as a different user:

admin@infinity:/home/vstubbs/git/OpenVoiceOS> ./scripts/br-patches.sh
can't find file to patch at input line 52
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 6a994bf9f58563e9f4c920037532c9590ccb78ed Mon Sep 17 00:00:00 2001
|From: Peter Steenbergen <info@j1nx.nl>
|Date: Mon, 13 Sep 2021 08:24:10 +0200
|Subject: [PATCH 1/1] Switch Python3.9.7 to Python3.8.12
|
|---
| ...e-the-build-of-pyc-files-conditional.patch | 14 +++----
| ...taddrinfo-configure-test-when-cross-.patch |  8 ++--
| ...re-to-disable-the-build-of-certain-e.patch | 20 +++++-----
| ...y-header-paths-for-cross-compilation.patch | 12 +++---
| ...ook-in-usr-lib-termcap-for-libraries.patch |  8 ++--
| .../0006-Don-t-add-multiarch-paths.patch      |  8 ++--
| .../0007-Abort-on-failed-module-build.patch   |  8 ++--
| .../0008-Serial-ioctl-workaround.patch        |  6 +--
| ...e-shebang-of-Python-scripts-for-cros.patch |  4 +-
| ...ig.sh.in-ensure-sed-invocations-only.patch |  4 +-
| ...locale-and-set-to-default-when-addin.patch | 14 +++----
| ...Add-importlib-fix-for-PEP-3147-issue.patch | 12 +++---
| ...-disable-installation-of-test-module.patch | 29 ++++++--------
| .../0014-Add-an-option-to-disable-pydoc.patch | 28 ++++++-------
| ...015-Add-an-option-to-disable-lib2to3.patch | 24 +++++------
| ...option-to-disable-the-sqlite3-module.patch | 16 ++++----
| ...d-an-option-to-disable-the-tk-module.patch | 20 +++++-----
| ...-option-to-disable-the-curses-module.patch | 26 ++++++------
| .../0019-Add-an-option-to-disable-expat.patch | 20 +++++-----
| ...-Add-an-option-to-disable-CJK-codecs.patch |  8 ++--
| .../0021-Add-an-option-to-disable-NIS.patch   |  8 ++--
| ...Add-an-option-to-disable-unicodedata.patch |  8 ++--
| .../0023-Add-an-option-to-disable-IDLE.patch  | 20 +++++-----
| ...024-Add-an-option-to-disable-decimal.patch | 16 ++++----
| ...on-to-disable-the-ossaudiodev-module.patch |  8 ++--
| ...an-option-to-disable-openssl-support.patch |  8 ++--
| ...ption-to-disable-the-readline-module.patch |  8 ++--
| ...to-disable-zlib-bzip2-and-xz-modules.patch |  8 ++--
| ...thon-config.sh-don-t-reassign-prefix.patch |  4 +-
| ...-Fix-cross-compiling-the-uuid-module.patch |  8 ++--
| ...Add-an-option-to-disable-uuid-module.patch |  8 ++--
| ...-fix-building-on-older-distributions.patch |  4 +-
| ...up-CC-print-multiarch-output-for-mus.patch |  8 ++--
| ...-ng-doesn-t-set-errno-when-encryptio.patch | 40 -------------------
| ...teration-error-in-_ExecutorManagerTh.patch | 29 --------------
| package/python3/python3.hash                  |  5 +--
| package/python3/python3.mk                    |  4 +-
| 37 files changed, 202 insertions(+), 281 deletions(-)
| delete mode 100644 package/python3/0034-lib-crypt-uClibc-ng-doesn-t-set-errno-when-encryptio.patch
| delete mode 100644 package/python3/0036-Fix-dictionary-iteration-error-in-_ExecutorManagerTh.patch
|
|diff --git a/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch b/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
|index 44902725d7..1010d08d53 100644
|--- a/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
|+++ b/package/python3/0001-Make-the-build-of-pyc-files-conditional.patch
--------------------------
File to patch: ^C
admin@infinity:/home/vstubbs/git/OpenVoiceOS>

denics, 2023-02-04:
Hi @vvstubbs do you still have this issue?

generallynonsensical, 2023-04-15:
I have a similar issue. I have patched and repatched. Checked my versions of scipy. I just don't know how to progress at this point. 

`patching file package/libabseil-cpp/libabseil-cpp.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/python3/python3.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 180 with fuzz 1 (offset -10 lines).
patching file package/python-pip/python-pip.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/python-gpiozero/Config.in
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 1 with fuzz 1.
The next patch would create the file package/mimic/0001-Disable-Werror.patch,
which already exists!  Assume -R? [n] y
patching file package/mimic/0001-Disable-Werror.patch
The next patch would create the file package/mimic/0002-Fix-pulseaudio-library-linking.patch,
which already exists!  Assume -R? [n] y
patching file package/mimic/0002-Fix-pulseaudio-library-linking.patch
patching file package/mimic/Config.in
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/mimic/mimic.hash
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/mimic/mimic.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/alsa-plugins/alsa-plugins.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/pulseaudio/pulseaudio.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
Hunk #1 succeeded at 159 (offset -3 lines).
patching file package/shairport-sync/shairport-sync.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/rpi-firmware/rpi-firmware.hash
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/rpi-firmware/rpi-firmware.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/qt5/qt5webengine/qt5webengine.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/vlc/vlc.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/cmake/cmake.mk
Reversed (or previously applied) patch detected!  Assume -R? [n] y
patching file package/brcmfmac_sdio-firmware-rpi/brcmfmac_sdio-firmware-rpi.hash
patching file package/brcmfmac_sdio-firmware-rpi/brcmfmac_sdio-firmware-rpi.mk
patching file package/qt5/qt5webengine/qt5webengine.mk
can't find file to patch at input line 15
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 18de31751accdc76b93c578435643ea487069fe3 Mon Sep 17 00:00:00 2001
|From: j1nx <p.steenbergen@j1nx.nl>
|Date: Fri, 7 Apr 2023 12:30:29 +0200
|Subject: [PATCH 1/1] Bump python-scipy
|
|---
| package/python-scipy/python-scipy.hash |  2 +-
| package/python-scipy/python-scipy.mk   | 13 +++++++++++--
| 2 files changed, 12 insertions(+), 3 deletions(-)
|
|diff --git a/package/python-scipy/python-scipy.hash b/package/python-scipy/python-scipy.hash
|index 0274679c43..453e89928f 100644
|--- a/package/python-scipy/python-scipy.hash
|+++ b/package/python-scipy/python-scipy.hash
--------------------------
File to patch: '

I have no clue how to choose which file to patch. I'm very new to this.

j1nx, 2023-12-29:
no longer needed

closed 2023-12-29 by j1nx