OpenVoiceOS/ovos-buildroot #114: Upstream changes to official Buildroot

tpetazzoni, 2022-08-18:
Hello! Great work with OpenVoiceOS! I am one of the maintainers of upstream Buildroot. You have lots of interesting packages that would make a lot of sense to have in upstream Buildroot. Have you considered contributing at least some of your changes to upstream Buildroot?

j1nx, 2022-08-19:
Hi Thomas! Nice to see you paying us a visit.

Short answer to your question; Absolutely.

Longer answer: I am still learning on the spot here and took some shortcuts here and there to get us what we wanted which still need to be fixed. 

We will get to it eventually and then we will look into sending patches over at the mailing list. (Also something I need to dig into / learn).

In the meantime if you or anyone of the other guru's I always see at the mailing list see something that you want. Ping me and I will see what I can do or you are more then welcome to just grab it. I will see it flying by upstream and learn a lot from it.

tpetazzoni, 2022-08-19:
Thanks for your feedback! Pretty much all the new packages that you have would be relevant for submission to upstream.

BTW, feel free to close this issue, because it isn't really a specific issue that can be considered "closed" at some point. It was merely a way of contacting you and saying that the upstream Buildroot project is interested in seeing your contributions.

closed 2024-01-24 by j1nx