OpenVoiceOS/ovos-buildroot #119: Can't SSH

ghamauricio, 2022-10-31:
Hello all.

I think I'm having trouble with my screens (one might be too high a resolution - 4k, seems to be utterly unresponsive - and the other is just plain problematic - had problems with it kinda resetting and crashing the pi in the past), so I'm trying to SSH into the Pi with no sucess.

Tried mycroft:mycroft, pi:mycroft, pi:raspberry, mycroft:raspberry with no sucess.

What could be wrong?

NeonDaniel, 2022-11-01:
Which image did you install? One from [openvoiceos.com](https://openvoiceos.com/) or one linked in the [Matrix chat](https://matrix.to/#/#openvoiceos-development:matrix.org)?

j1nx, 2023-12-29:
The current user/passwd combo for this image is ovos/ovos

closed 2023-12-29 by j1nx