OpenVoiceOS/ovos-buildroot #124: GUI build error

builderjer, 2022-12-17:
I am trying to build this for rpi4-gui.  My host is ubuntu server 22.04
I cant get past this error
```
-- Performing Test HAVE_DATE_TIME - Success
fatal: HEAD does not point to a branch
CMake Deprecation Warning at /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/share/ECM/modules/ECMQMLModules.cmake:7 (message):
  ECMQMLModules.cmake is deprecated since 5.88.0, please use
  ECMFindQmlModule.cmake instead
Call Stack (most recent call first):
  CMakeLists.txt:40 (include)

CMake Error at /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/cmake/Qca-qt5/Qca-qt5Targets.cmake:73 (message):
  The imported target "qca-qt5" references the file

     "usr/lib/libqca-qt5.so.2.3.4"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/cmake/Qca-qt5/Qca-qt5Targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/cmake/Qca-qt5/Qca-qt5Config.cmake:38 (include)
  CMakeLists.txt:62 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/jbrodie/software/ovos-buildroot/buildroot/output/build/kf5-kdeconnect-v22.04.1/buildroot-build/CMakeFiles/CMakeOutput.log".
make[2]: *** [package/pkg-generic.mk:283: /home/jbrodie/software/ovos-buildroot/buildroot/output/build/kf5-kdeconnect-v22.04.1/.stamp_configured] Error 1
make[1]: *** [Makefile:84: _all] Error 2
make[1]: Leaving directory '/home/jbrodie/software/ovos-buildroot/buildroot'
rsync -ah --progress buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
sending incremental file list
rsync: [sender] link_stat "/home/jbrodie/software/ovos-buildroot/buildroot/output/images/sdcard.img" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]
make: *** [Makefile:26: rpi4_64-gui] Error 23
```

but I have that file I think 
```
ls -l /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/libqca-qt5.so*
lrwxrwxrwx 1 jbrodie jbrodie      15 Dec 15 18:47 /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/libqca-qt5.so -> libqca-qt5.so.2
lrwxrwxrwx 1 jbrodie jbrodie      19 Dec 15 18:47 /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/libqca-qt5.so.2 -> libqca-qt5.so.2.3.4
-rwxr-xr-x 1 jbrodie jbrodie 1528240 Dec 15 18:47 /home/jbrodie/software/ovos-buildroot/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/usr/lib/libqca-qt5.so.2.3.4
```

I would love to get this to build once, so I can make my own changes.  Any help would be greatly appreciated

builderjer, 2022-12-18:
I found a workaround [here](https://github.com/OpenVoiceOS/ovos-buildroot/issues/116#issuecomment-1255306774) and that got me past that.  Used the same for a couple more things. And then used the methods [here](https://matrix.to/#/!EOygeDJPfJQOfNacqH:matrix.org/$DkPBH0Smw1QepsqK5s3MHGSwBB6yHbennDiqF34jZLE?via=matrix.org&via=friends.famedly.de&via=kde.org) on a few packages and not have a completed image.  It is WITHOUT kdeconnect though.  Any help getting that to work would be awesome as I like the feature.

j1nx, 2023-12-29:
Fixed / no longer needed.

closed 2023-12-29 by j1nx