OpenVoiceOS/ovos-buildroot #132: [BUG] Wrong checksum for ovos-stt-server-plugin

atd, 2023-02-05:
When building the image from the last commit in `develop` I get an error regarding the checksum of `ovos-stt-server-plugin`

Maybe the dependency has not been updated to the last version? Or something weird is happening?

```
>>> python-ovos-stt-server-plugin c8d7dea5a6875221f152828629846979f52e6d05 Downloading
wget --passive-ftp -nd -t 3 -O '/home/ovos/OpenVoiceOS/buildroot/output/build/.python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz.0foFcn/output' 'https://github.com/OpenVoiceOS/ovos-stt-server-plugin/archive/c8d7dea5a6875221f152828629846979f52e6d05/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz'
--2023-02-05 23:07:59--  https://github.com/OpenVoiceOS/ovos-stt-server-plugin/archive/c8d7dea5a6875221f152828629846979f52e6d05/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://github.com/OpenVoiceOS/ovos-stt-plugin-server/archive/c8d7dea5a6875221f152828629846979f52e6d05/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz [following]
--2023-02-05 23:08:00--  https://github.com/OpenVoiceOS/ovos-stt-plugin-server/archive/c8d7dea5a6875221f152828629846979f52e6d05/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/OpenVoiceOS/ovos-stt-plugin-server/tar.gz/c8d7dea5a6875221f152828629846979f52e6d05 [following]
--2023-02-05 23:08:00--  https://codeload.github.com/OpenVoiceOS/ovos-stt-plugin-server/tar.gz/c8d7dea5a6875221f152828629846979f52e6d05
Resolving codeload.github.com (codeload.github.com)... 140.82.121.9
Connecting to codeload.github.com (codeload.github.com)|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/home/ovos/OpenVoiceOS/buildroot/output/build/.python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz.0foFcn/output'

     0K ...                                                    2.50M=0.001s

2023-02-05 23:08:00 (2.50 MB/s) - '/home/ovos/OpenVoiceOS/buildroot/output/build/.python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz.0foFcn/output' saved [3537]

ERROR: python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz has wrong sha256 hash:
ERROR: expected: 803f37b5be237c62e1afae696935fd2e58bc0648a0a8906f7855f42d880b2476
ERROR: got     : 4a5136c3fe8ad6524f3bcbbf87f08f1369ceade243972e1c942294e007dc1370
ERROR: Incomplete download, or man-in-the-middle (MITM) attack
wget --passive-ftp -nd -t 3 -O '/home/ovos/OpenVoiceOS/buildroot/output/build/.python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz.Sa1seK/output' 'http://sources.buildroot.net/python-ovos-stt-server-plugin/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz'
--2023-02-05 23:08:00--  http://sources.buildroot.net/python-ovos-stt-server-plugin/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 172.67.72.56, 104.26.1.37, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|172.67.72.56|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-02-05 23:08:00 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/home/ovos/OpenVoiceOS/buildroot/output/build/.python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz.osfjHR/output' 'http://sources.buildroot.net/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz'
--2023-02-05 23:08:00--  http://sources.buildroot.net/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.26.1.37, 172.67.72.56, 104.26.0.37, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.26.1.37|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-02-05 23:08:00 ERROR 404: Not Found.

make[1]: *** [package/pkg-generic.mk:189: /home/ovos/OpenVoiceOS/buildroot/output/build/python-ovos-stt-server-plugin-c8d7dea5a6875221f152828629846979f52e6d05/.stamp_downloaded] Error 1
make[1]: Leaving directory '/home/ovos/OpenVoiceOS/buildroot'
rsync -ah --progress buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
sending incremental file list
rsync: [sender] link_stat "/home/ovos/OpenVoiceOS/buildroot/output/images/sdcard.img" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
make: *** [Makefile:26: rpi4_64-gui] Error 23

```

dickorydock, 2023-02-25:
I am experiencing the same checksum issue for rpi3. Some of the issues I ran into trying to compile before have been fixed, and this checksum one is the first one I haven't been able to get through yet.

j1nx, 2023-03-14:
I thought I fixed this one. Will double check and change accordingly.

j1nx, 2023-12-29:
fixed (and also no longer needed)

closed 2023-12-29 by j1nx