OpenVoiceOS/ovos-buildroot #137: [BUG] when launching phal : ntpd[1065]: must be run as root, not uid 1000

denics, 2023-03-04:
**OpenVoiceOS specifications (please complete the following information):**
 - Device: Mark2-Devkit
 - Image release: 20230117

**Describe the bug**
When launching the systemd phal component we have this:

```
mycroft@OpenVoiceOS-e383a2:~ $ tail -f /var/log/mycroft/phal.error.log
Connection Refused. Is Messagebus Service running?
Exception closing websocket at ws://0.0.0.0:8181/core: [Errno 111] Connection refused
Message Bus Client will reconnect in 10.0 seconds.
 * Running on http://0.0.0.0:36536/ (Press CTRL+C to quit)
/bin/sh: line 1: service: command not found
 4 Mar 19:42:04 ntpd[937]: must be run as root, not uid 1000
/bin/sh: line 1: service: command not found
/bin/sh: line 1: service: command not found
 4 Mar 19:42:08 ntpd[1065]: must be run as root, not uid 1000
/bin/sh: line 1: service: command not found
```

j1nx, 2023-03-14:
Thanks, I have also seen this. I will look into it. Switching it over to use timesyncd of systemd. That one will allow for better user control.

j1nx, 2024-01-27:
For this to resolve we need this PR merged first;

https://github.com/OpenVoiceOS/ovos-utils/pull/200#issuecomment-1913171531