OpenVoiceOS/ovos-buildroot #141: EOF error on Ubuntu 22.04

JoergZ2, 2023-03-23:
Got this error when calling make rpi4_64-gui:
```
jz@jz-desktop:~/OpenVoiceOS$ make rpi4_64-gui
config rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external "rpi4_64-gui_defconfig"
make[1]: Verzeichnis „/home/jz/OpenVoiceOS/buildroot“ wird betreten
#
# configuration written to /home/jz/OpenVoiceOS/buildroot/.config
#
make[1]: Verzeichnis „/home/jz/OpenVoiceOS/buildroot“ wird verlassen
build rpi4_64-gui
make -C buildroot BR2_EXTERNAL=../buildroot-external 2>&1 | tee logs/buildroot_rpi4_64-gui_output.txt
make[1]: Verzeichnis „/home/jz/OpenVoiceOS/buildroot“ wird betreten
/usr/bin/make -j1 O=/home/jz/OpenVoiceOS/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
>>> host-skeleton  Extracting
>>> host-skeleton  Patching
>>> host-skeleton  Configuring
>>> host-skeleton  Building
>>> host-skeleton  Installing to host directory
/bin/bash: -c: Zeile 1: Dateiende beim Suchen nach »"« erreicht.
/bin/bash: -c: Zeile 2: Syntaxfehler: Unerwartetes Dateiende.
make[2]: *** [package/pkg-generic.mk:189: /home/jz/OpenVoiceOS/buildroot/output/build/host-ccache-3.7.12/.stamp_downloaded] Fehler 2
make[1]: *** [Makefile:84: _all] Fehler 2
make[1]: Verzeichnis „/home/jz/OpenVoiceOS/buildroot“ wird verlassen
rsync -ah --progress buildroot/output/images/sdcard.img release/OpenVoiceOS_rpi4_64-gui.img
sending incremental file list
rsync: [sender] link_stat "/home/jz/OpenVoiceOS/buildroot/output/images/sdcard.img" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
make: *** [Makefile:26: rpi4_64-gui] Fehler 23
```

j1nx, 2023-03-23:
I need to dig into this as we do not touch the host-skeleton package at all, I guess you run into an upstream bug.

Will see what I can do / can find and report back as soon as I have found something.

j1nx, 2023-12-29:
Guess this is fixed with the new build system as well. If not, please let me know.

closed 2023-12-29 by j1nx