OpenVoiceOS/ovos-buildroot #173: [BUG] all fallback answer

denics, 2023-05-23:
**OpenVoiceOS specifications (please complete the following information):**
 - Device: Mark2-Devkit
 - Image release: january 2023

**Describe the bug**
If you install _skill-ovos-fallback-chatgpt_ along another fallback skill, both fallback will answer

**To Reproduce**
Steps to reproduce the behavior:
1. Install skill-ovos-fallback-unknown
2. Install skill-ovos-fallback-chatgpt
3. Ask something weird (it happen a lot when not using English)
4. Both fallback skills answer

**Expected behavior**
The first fallback that answer should close the dialogue.

**Additional context**
Chatgpt takes a long time to answer, probably this is the reason why both fallback kick in.

j1nx, 2023-12-29:
Will look into this issue. as it might be related to; https://github.com/OpenVoiceOS/ovos-core/issues/383

denics, 2023-12-30:
It seems more an issue with the timeout I think, I will try to find the reference in the matrix chat, we discussed this once with @JarbasAl

JarbasAl, 2023-12-30:
with latest gpt skill this should be fixed