OpenVoiceOS/ovos-buildroot #191: [BUG] Doesn't boot in Mark II: blank screen

atd, 2023-09-10:
I just built the image with the [current code](https://github.com/OpenVoiceOS/ovos-buildroot/commit/0035e2ffbf3d067e73078e56eb28d67f83f43300). I had no problems building it, but I couldn't follow instructions at https://openvoiceos.github.io/community-docs/images_buildroot/

* I cannot patch, I get `./scripts/br-patches.sh: line 5: buildroot-patches/*: No such file or directory`
* Target images have changed. I am using `rpi4_64`

I use latest version of `rpi-imager` to prepare the USB stick

When I boot the Mark II, I can see the boot menu of the rpi-imager. It says it boots OVOS in the first slot. Last message I see is:

```
Booting "OpenVoiceOS - Slot A"
```

After that, only a blank screen

j1nx, 2023-09-11:
The buildroot patches are no longer needed. I am working on the changed documentation, however not yet ready nor merged;
https://github.com/OpenVoiceOS/ovos-buildroot/tree/documentation

We are currently in process of migration our architecture to make used of a containerized environment. It allows us to seperate the Host OS development from the OVOS framework development. So in short, you just catched us at the wrong time.

There is work ongoing behind the scenes to change the GUI framework to allows it to be ran in a cotainer which is not yet ready and the reason the GUI is not fully included within this new structure.

When the new sourcecode and images are ready to be tested, it will be communicated. However fee lfree to look around the current codebase to see where things are at. If you connect a LAN cable, most services shpould be pulled in and started. However it still might require certain things to be configured manually here and there.

j1nx, 2023-12-29:
Found the culprit of the non booting of images. I was missing a folder for teh system to properly mount the EFI partition.

Fixed now

closed 2023-12-29 by j1nx