OpenVoiceOS/ovos-buildroot #202: [BUG] Build problem, ovos-buildroot with x86_64-base

DetergentW10, 2023-12-16:
When building the image from develop I get an error:

` 57900K .......... .......... .......... .......... .....       148K=3m55s

2023-12-16 23:14:20 (246 KB/s) - ‘/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/build/.deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz.YwQkIq/output’ saved [59336129]

    Updating git repository `https://github.com/aldanor/hdf5-rust/`
error: failed to sync

Caused by:
  failed to load pkg lockfile

Caused by:
  failed to get `hdf5` as a dependency of package `deep_filter v0.4.1-pre (/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/build/.deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz.YwQkIq/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800/libDF)`

Caused by:
  failed to load source for dependency `hdf5`

Caused by:
  Unable to update https://github.com/aldanor/hdf5-rust/?rev=refs/pull/216/head#fd323b3d

Caused by:
  object not found - no match for id (fd323b3d88d9f4eb339c5ccd1f3127a2367da3c8); class=Odb (9); code=NotFound (-3)
wget --passive-ftp -nd -t 3 -O '/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/build/.deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz.MxD8kE/output' 'https://sources.buildroot.net/deepfilternet-ladspa/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz' 
--2023-12-16 23:14:29--  https://sources.buildroot.net/deepfilternet-ladspa/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 2606:4700:20::ac43:4838, 2606:4700:20::681a:125, 2606:4700:20::681a:25, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|2606:4700:20::ac43:4838|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-12-16 23:14:29 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/build/.deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz.LWGPqw/output' 'https://sources.buildroot.net/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz' 
--2023-12-16 23:14:29--  https://sources.buildroot.net/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 2606:4700:20::681a:25, 2606:4700:20::681a:125, 2606:4700:20::ac43:4838, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|2606:4700:20::681a:25|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-12-16 23:14:30 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/build/.deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz.Fp7KZB/output' 'https://github.com/skiffos/mirror/raw/master/deepfilternet-ladspa/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz' 
--2023-12-16 23:14:30--  https://github.com/skiffos/mirror/raw/master/deepfilternet-ladspa/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800.tar.gz
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-12-16 23:14:31 ERROR 404: Not Found.

make[1]: *** [package/pkg-generic.mk:179: /media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/build/deepfilternet-ladspa-27b5f07be6caba1abec7907fcca89ba906732800/.stamp_downloaded] Error 1
make[1]: Leaving directory '/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot'
rsync -ah --progress buildroot/output/images/disk.img release/OpenVoiceOS_x86_64-base_20231216_2210.img
sending incremental file list
rsync: [sender] link_stat "/media/2945e18b-2c55-4874-93df-7622d24d7eb6/OpenVoiceOS/ovos-buildroot/buildroot/output/images/disk.img" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
make: *** [Makefile:26: x86_64-base] Error 23
`
[buildroot_x86_64-base_output.txt](https://github.com/OpenVoiceOS/ovos-buildroot/files/13694112/buildroot_x86_64-base_output.txt)

j1nx, 2023-12-17:
Apologies! I have not been updating the x86-64 defconfig for the new container based system as while working on it i focussed on the rpi boards and the ova system for testing.

I will try to get that defconfig up to date somewhere this week.

DetergentW10, 2023-12-17:
Thank you very much!

j1nx, 2024-01-24:
Apologies. Ended up into a little rabbit whole while migrating from systemd-networkd towards NetworkManager.

Will look into getting the rpi3, ova and x84 defconfigs into shape.

j1nx, 2024-01-26:
Should be fixed now with;
https://github.com/OpenVoiceOS/ovos-buildroot/commit/5e7fd2e185ba723c1f5a50c871d5e4f9ceb78d06

Haven't booted it as of yet. Needs extensive testing going forward.

Closing this issue in favor of any newly opened issue for bugs that surface during testing.