OpenVoiceOS/ovos-buildroot #218: [BUG] unable to build in LXC contianer, tried ubuntu server 20.04 and 22.04 and debian 12

cooljimy84, 2024-02-16:
I'm trying to build in a lxc container on proxmox, i've tried the above containers but always get hit with the following error, when building either rpi3 or rpi4
```
make[3]: Leaving directory '/home/builder/ovos-buildroot/buildroot/output/build/host-zstd-1.5.4/lib'
CC obj/conf_c8e3a4bd2445990d7a524270e236ecc4/datagen.o
CC obj/conf_c8e3a4bd2445990d7a524270e236ecc4/dibio.o
CC obj/conf_c8e3a4bd2445990d7a524270e236ecc4/fileio.o
CC obj/conf_c8e3a4bd2445990d7a524270e236ecc4/fileio_asyncio.o
fileio.c: In function 'FIO_compressLzmaFrame':
fileio.c:1176:22: error: 'Bytef' undeclared (first use in this function)
 1176 |     strm.next_out = (Bytef*)writeJob->buffer;
      |                      ^~~~~
fileio.c:1176:22: note: each undeclared identifier is reported only once for each function it appears in
fileio.c:1176:28: error: expected expression before ')' token
 1176 |     strm.next_out = (Bytef*)writeJob->buffer;
      |                            ^
fileio.c:1177:23: error: 'uInt' undeclared (first use in this function)
 1177 |     strm.avail_out = (uInt)writeJob->bufferSize;
      |                       ^~~~
fileio.c:1177:28: error: expected ';' before 'writeJob'
 1177 |     strm.avail_out = (uInt)writeJob->bufferSize;
      |                            ^~~~~~~~
      |                            ;
fileio.c:1204:40: error: expected expression before ')' token
 1204 |                 strm.next_out = (Bytef*)writeJob->buffer;
      |                                        ^
fileio.c: In function 'FIO_decompressLzmaFrame':
fileio.c:2319:22: error: 'Bytef' undeclared (first use in this function)
 2319 |     strm.next_out = (Bytef*)writeJob->buffer;
      |                      ^~~~~
fileio.c:2319:28: error: expected expression before ')' token
 2319 |     strm.next_out = (Bytef*)writeJob->buffer;
      |                            ^
fileio.c:2320:23: error: 'uInt' undeclared (first use in this function)
 2320 |     strm.avail_out = (uInt)writeJob->bufferSize;
      |                       ^~~~
fileio.c:2320:28: error: expected ';' before 'writeJob'
 2320 |     strm.avail_out = (uInt)writeJob->bufferSize;
      |                            ^~~~~~~~
      |                            ;
fileio.c:2348:40: error: expected expression before ')' token
 2348 |                 strm.next_out = (Bytef*)writeJob->buffer;
      |                                        ^
make[4]: *** [Makefile:330: obj/conf_c8e3a4bd2445990d7a524270e236ecc4/fileio.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:150: zstd] Error 2
make[3]: Leaving directory '/home/builder/ovos-buildroot/buildroot/output/build/host-zstd-1.5.4/programs'
make[2]: *** [Makefile:67: zstd-release] Error 2
make[2]: Leaving directory '/home/builder/ovos-buildroot/buildroot/output/build/host-zstd-1.5.4'
make[1]: *** [package/pkg-generic.mk:283: /home/builder/ovos-buildroot/buildroot/output/build/host-zstd-1.5.4/.stamp_built] Error 2
make[1]: Leaving directory '/home/builder/ovos-buildroot/buildroot'
rsync -ah --progress buildroot/output/images/disk.img release/OpenVoiceOS_rpi4_64_20240216_1343.img
sending incremental file list
rsync: [sender] link_stat "/home/builder/ovos-buildroot/buildroot/output/images/disk.img" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
make: *** [Makefile:26: rpi4_64] Error 23
```
I had a quick google and made sure i have zstd installed

j1nx, 2024-02-17:
I have traced this error back to upstream zstd issue;
https://github.com/facebook/zstd/issues/3580

As we use version 1.5.4 within the 2023.02.x branch of buildroot, however the master branch upstream has a bump to 1.5.5 I guess the version bump has the fix implemented.

Could you give it another try with the latest code?

Easiest I guess would be to run a ```git pull --recurse-submodules``` ```make clean``` followed by ```make rpi4_64```

cooljimy84, 2024-02-19:
Seems to have got further than last time (only 10 minutes in tho)

mshelby, 2024-02-19:
Hey, can somebody unsubscribe me from all OpenVoiceOS threads?

I keep 'unsubscribing' via the email option but that just unsubscribes me
from that particular thread, not from all emails.

Thus, with every new thread topic I get a new email.

Mark Shelby

emphasize, 2024-02-19:
@mshelby [github.com/watching](http://github.com/watching)

closed 2024-02-20 by cooljimy84