OpenVoiceOS/ovos-buildroot #220: [BUG] No space left of device (28)

cooljimy84, 2024-02-20:
```

ovos/.local/share/containers/storage/vfs/dir/5f85ae614164fe3ab2aabcc133cdd70b94e3a65dc2d214315b30f9fe9e276233/usr/share/perl/5.36.0/Module/CoreList.pm
         32,768   3%   47.13kB/s    0:00:20  rsync: [receiver] write failed on "/home/builder/ovos-buildroot/buildroot/output/images/home/ovos/.local/share/containers/storage/vfs/dir/5f85ae614164fe3ab2aabcc133cdd70b94e3a65dc2d214315b30f9fe9e276233/usr/share/perl/5.36.0/Module/CoreList.pm": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(380) [receiver=3.2.7]

rsync: [sender] write error: Broken pipe (32)
make[1]: *** [Makefile:751: target-finalize] Error 11
make[1]: Leaving directory '/home/builder/ovos-buildroot/buildroot'
rsync -ah --progress buildroot/output/images/disk.img release/OpenVoiceOS_rpi3_64_20240220_1519.img
sending incremental file list
rsync: [sender] link_stat "/home/builder/ovos-buildroot/buildroot/output/images/disk.img" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
make: *** [Makefile:26: rpi3_64] Error 23

```

```
builder@buildvm:~/ovos-buildroot$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            4.4G     0  4.4G   0% /dev
tmpfs           898M  816K  897M   1% /run
/dev/sda1       123G   68G   49G  59% /
tmpfs           4.4G  168K  4.4G   1% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           898M  6.8M  891M   1% /run/user/1000
/dev/loop0      6.6G  6.6G     0 100% /home/builder/ovos-buildroot/buildroot/output/images/home
```

j1nx, 2024-02-22:
You are using teh VFS driver on your system instead of the overlayfs.

I am currently working on building host podman/skopeo/jq tools to account for this. As a possible quickfix you could create/add the below file in the directory of your build host system

~/.config/containers/storage.conf

https://github.com/OpenVoiceOS/ovos-buildroot/blob/develop/buildroot-external/rootfs-overlay/home/ovos/.config/containers/storage.conf

cooljimy84, 2024-02-22:
I'll hold off making a pull request for the image creation "homefs.ext4" size as just making it 20GB is kinda over kill (and it seems to shrink it down to the sized used after it's finished copying)
Unless you want me to test and possibly change the other build options pi4, x86 and find a size that currently works ?